hultenvp/solis-sensor #6: Not working with two accoutns

richhalliwell, 2021-05-22:
When i try and use this for two separate accounts (mine and my neighbours) it doesn't seem to work. Occasionally it will work for one and not the other. I suspect it is attempting to use the same session for both and not creating two separate login sessions? Can you confirm my suspicions?

hultenvp, 2021-05-22:
@richhalliwell Yeah, I think your suspicion might be true. I'm not sure, but I might need to fetch http session only once and store it for further use. I'll try to create a test version for you. I'll ping you when done.

hultenvp, 2021-05-22:
@richhalliwell Please try the code here: https://github.com/hultenvp/solis-sensor/tree/6-not-working-with-2-accounts/custom_components/solis

hultenvp, 2021-06-30:
@richhalliwell: Any progress? Does the fix work?

hultenvp, 2021-09-30:
@richhalliwell : Gentle reminder. Does the fix work?

jdpl-ratel, 2022-02-16:
If something has changed in this topic and you can now run two instances?

hultenvp, 2022-02-17:
Each account specified is one instance.
Each instance only supports one Plant, but all inverters in that plant.