hultenvp/solis-sensor #8: Support for energy dashboard

hultenvp, 2021-08-29:
Add extra properties to support new HA energy dashboard (see https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics):
* Device class
* State class

FilipNaudts, 2021-09-22:
Can you give an example?

larsenglund, 2021-09-29:
When you want to att solar production to your energy config (Configuration -> Energy -> Solar Panels (ADD SOLAR PRODUCTION)) none of the solis sensors show up as a selectable sensor for "Solar production energy (kWh)". I think, after reading the documentation @hultenvp linked to, that for example the sensor "energytoday" needs to have it's state_class set to total_increasing to show up as a valid sensor. When the state class is set to total_increasing HA starts tracking statistics for that sensor.

larsenglund, 2021-09-29:
I tried changing the class but I can't find any documentation or exmple on how to do it so this is just a guess that didn't work..

```
class SolisSensor(Entity):
  """ Representation of a Solis sensor. """

  def __init__(self, inverter_name, sensor_type):
    # Initialize the sensor.
    self._inverter_name = inverter_name
    self._type = sensor_type
    self._measured = None
    # Properties
    self._icon = SENSOR_TYPES[sensor_type][2]
    self._name = self._inverter_name + ' ' + SENSOR_TYPES[sensor_type][0]
    self._state = None
    self._uom = SENSOR_TYPES[sensor_type][1]
    if (sensor_type == "energytoday"):
      _LOGGER.debug("Solis setting device and state class...")
      self._attr_device_class = sensor.DEVICE_CLASS_ENERGY
      self._attr_state_class = sensor.STATE_CLASS_TOTAL_INCREASING
```
Anyone know how to set the classes?

larsenglund, 2021-09-29:
The Tibber integration (which works as an Energy sensor source) seems to use a different approach to setting up the sensors with the correct device classes. I don't know how to modify this integration to use the same method though...
https://github.com/home-assistant/core/blob/dev/homeassistant/components/tibber/sensor.py

larsenglund, 2021-09-29:
Looking at some other [random project](https://github.com/cyberjunky/home-assistant-toon_smartmeter/blob/master/custom_components/toon_smartmeter/sensor.py) they use `self._attr_device_class` in `__init__` to assign a class so I guess my code should have worked? Hmmz..

hultenvp, 2021-09-30:
Hey @larsenglund , nice to see interest in my sensor and interest to contribute. I can confirm all your findings. Attr needs to be set to toal increasing and nope, setting self._attr_device_class does indeed also not work. 
I'm assuming (did not check actual code of the energy dashboard) that it expects sensors to derive from the sensor base class and not the entity base class. 
Seems to be the way forward anyhow for sensors, so now rewriting solis to derive from sensor base class. Hopefully I get it working soon, I'll keep you posted

hultenvp, 2021-09-30:
Hey @larsenglund, I managed to get things working. Moving to SensorEntity base class did the trick.

Could you verify it also works for you? Code is in 
https://github.com/hultenvp/solis-sensor/tree/8-Support-for-energy-dashboard 
branch

Cheers,
Peter

larsenglund, 2021-09-30:
Horray! Freakishly fast update there @hultenvp! I could select `solis_inverter_energy_today` as my production source in the Energy config now. I assume it will work but HA seems to want some time to collect data from it before showing anything in the GUI, I'll report back in a couple of hours :)
First it failed but that was just because I installed it using HACS and the `custom_components.json` in the branch still pointed to the main repo 😅

hultenvp, 2021-09-30:
I assume you use custom repositories in HACS? Still need to look into how to register with HACS someday.
Anyway. All worked fine, felt confident enough to push to master. I also used 'solis_inverter_energy_today'. Started showing graphs after 2 updates of the value. Took about half an hour as it is now heavily overcast.
Could you perhaps verify it works via the HACS route?

larsenglund, 2021-09-30:
Yes, I used a custom repository to install it. 
Hmm, the graph still says _"There is no data to show. It can take up to 2 hours for new data to arrive after you configure your energy dashboard."_ And the energy distribution graphic still shows zero for the panels even though the sensor clearly has a value.
[image]
[image]

larsenglund, 2021-09-30:
Nevermind, it just started working! It's very overcast here today so I guess that's why.

larsenglund, 2021-09-30:
I can confirm that installing via HACS works, it showed that an update was available now that you pushed to master and the update installed fine.

hultenvp, 2021-09-30:
@FilipNaudts: Support has been added to master, please read readme.md there for more details.

hultenvp, 2021-09-30:
@larsenglund : THanks for confirming and glad to see it also works at your side. Closing this issue as resolved.

closed 2021-09-30 by hultenvp

FilipNaudts, 2021-10-01:
Nice work guys!