hultenvp/solis-sensor #16: Battery status as a sensor

thomclae33, 2021-09-30:
The solis sensor integration works like a charm 🙌.
I wondered if it would be possible to add the Battery Charge Percent as a sensor as well?

hultenvp, 2021-10-01:
Hey @thomclae33: I'm unsure what you mean with battery charge percent? I've looked on my inverter dashboard (m.ginlong.com), but I don't see any reference to battery charge.
Can you share some details?

thomclae33, 2021-10-01:
Yes, sure. When you have a home battery installed and connected to a Solis inverter, you can see its battery percentage on m.ginlong.com.  This Ginlong-scraper has defined it as 'Battery_charge_percent': https://github.com/dkruyt/ginlong-scraper/blob/master/ginlong-scraper.py. Not sure how to get that data and how to get it into Home Assistant though.
[image]

lorrainem77, 2021-10-02:
I came here to write the exact same as @thomclae33, this is a great integration, and hoped there could be some way to add battery detail.

thomclae33, 2021-10-02:
Found some more that might be helpful:
[image]

hultenvp, 2021-10-02:
Ah, the last comment is definitely helpful. With the key it should be easy to add. 
Hopefully I have some time coming week to add it.

thomclae33, 2021-10-02:
Below is all the battery data I could find.
'Remaining Battery  Capacity (SoC)', 'Total Energy Charged' and 'Total Energy Discharged' are probably the most interesting. I couldn't find much useful in the [Home Assistant docs.](https://www.home-assistant.io/docs/energy/battery/).. 

```JSON
"realTimeDataBattery": [{
                    "key": "1ff",
                    "name": "Battery Status",
                    "value": "Discharge"
                }, {
                    "key": "1cr",
                    "name": "Battery Voltage",
                    "unit": "V",
                    "value": "48.10"
                }, {
                    "key": "1cs",
                    "name": "Battery  Current",
                    "unit": "A",
                    "value": "5.00"
                }, {
                    "key": "1ct",
                    "name": "Battery  Power",
                    "unit": "W",
                    "value": "240"
                }, {
                    "key": "1cv",
                    "name": "Remaining Battery  Capacity (SoC)",
                    "unit": "%",
                    "value": "13"
                }, {
                    "key": "2tm",
                    "name": "SOH",
                    "value": "100"
                }, {
                    "key": "1cx",
                    "name": "Total Energy Charged",
                    "unit": "kWh",
                    "value": "4"
                }, {
                    "key": "1cy",
                    "name": "Total Energy Discharged",
                    "unit": "kWh",
                    "value": "10"
                }, {
                    "key": "2vc",
                    "name": "Energy Charged yesterday",
                    "unit": "kWh",
                    "value": "1.40"
                }, {
                    "key": "2vd",
                    "name": "Energy Discharged yesterday",
                    "unit": "kWh",
                    "value": "5.20"
                }, {
                    "key": "1cz",
                    "name": "Daily Energy Charged",
                    "unit": "kWh",
                    "value": "0.80"
                }, {
                    "key": "1da",
                    "name": "Daily Energy Discharged",
                    "unit": "kWh",
                    "value": "3.20"
                }, {
                    "key": "1db",
                    "name": "Monthly Energy Charged",
                    "unit": "kWh",
                    "value": "2.2"
                }, {
                    "key": "1dc",
                    "name": "Monthly Energy Discharged",
                    "unit": "kWh",
                    "value": "8.39"
                }, {
                    "key": "1dd",
                    "name": "Annual Energy Charged",
                    "unit": "kWh",
                    "value": "4.69"
                }, {
                    "key": "1de",
                    "name": "Annual Energy Discharged",
                    "unit": "kWh",
                    "value": "10.2"
                }],
```

hultenvp, 2021-10-03:
Thanks for the detailed info @thomclae33.
There link you provided indeed gives little info other than that it's supposed to be sensor data. That's in principle easy to make.
However, in order to make the right design decisions i'd like to know how the battery info is displayed on the portal website.
Is the battery info part of the inverter device or is the battery listed as a separate device?
In case of the latter I need to make some changes to allow simultaneously scraping of multiple devices each with their own serial. Probably also means some breaking changes to the configuration then. 
If it's part of the same device then it's a few lines of code.

thomclae33, 2021-10-03:
The battery is part of the inverter; So no other devices are listed on the portal.

hultenvp, 2021-10-04:
@thomclae33 , @lorrainem77 : Got a version for you to test:
https://github.com/hultenvp/solis-sensor/tree/16-Battery_status_as_a_sensor
readme.md contains new sensor names:
      batcapacityremaining:  
      battotalenergycharged:
      battotalenergydischarged:

Hope it works, otherwise I need some more details from the scrape

lorrainem77, 2021-10-04:
Thanks for this @hultenvp, I've updated with the latest files but the battery sensors are not appearing for me. I edited the platform2_portal file, as it looks like some lines were copied and pasted in error (see line 11). Is it possible there might be other issues with this file?

hultenvp, 2021-10-05:
Yeah, something went horribly wrong in copy-pasting (still have to properly set up my dev env since migrating my Home Assistant instance). Should be fixed now. I'll double check later today.

thomclae33, 2021-10-05:
It looks like it works, @hultenvp ! Nice!
It is not showing up in the home battery selection (under 'energy'). I assume because[ long-term statistics](https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics) are not enabled yet for these sensors. I couldn't immediately find which ones are lacking.

lorrainem77, 2021-10-05:
Working perfectly for me too, 3 sensors have appeared and allowed me to add the charge and discharge to the Battery section on the energy dashboard - great work @hultenvp :)

hultenvp, 2021-10-05:
Nice to hear it works! Also nice that it does show up on the dashboard.
I implemented the 3 sensors that @thomclae33 suggested. Any more battery sensors you'd like to see added? Now's the time to ask. ;-)
Otherwise I create a PR and make a new release.