hultenvp/solis-sensor #25: Other m.ginlong.com data?

hawkeye100, 2021-10-29:
I notice that when 'exporting' from the device page in m.ginlong.com the resultant spreadsheet has more fields/attributes than apparently exposed by the API in this project's code.

Is it possible to access some of this other data by modifying this code?

I am interested in :
- Power Grid total power (W)
- Total Consumption power (W)

(unless these can be obtained through calculation from the existing data, though I could not see how to do that)

Maybe it is feasible to substitute some of the existing fields/attributes with these to save extending the code - I do not pretend to know how.

It would also be helpful to have the last update date/time available - perhaps as an entity to display on a card in HA. Is that feasible?

TIA

hultenvp, 2021-10-30:
Hi @hawkeye100 

Good timing, I was just adding a few others as part of #23 .
I can take these along, I have the keys.

Last update data/time should also be possible. Also considering a new binary sensor to indicate logged in y/n, but maybe that will follow later as it will be a separate sensor.

I'll ping you when done, I hope you're able to help testing if they work.

hultenvp, 2021-10-30:
Hey @hawkeye100 

I added the 2 power values.
Date-time stamp I added as a separate attribute as each value from the solis site is treated as a separate sensor that is only updated if something changes. So a global timestamp sensor does not make sense.

[image]

You can find the changes in https://github.com/hultenvp/solis-sensor/tree/23-add-more-sensors-for-battery
Can you confirm you can receive the data? Readme contains the config elements you can use.

Cheers

hawkeye100, 2021-10-31:
Many thanks for the update - looks like what I need.

Testing:
I stopped the server, then

I have replaced the files in the custom_components/solis folder with those from the 23-add-more-sensors-for-battery code.

I added the following to configuration.yaml under sensors for this component:
      battotalenergycharged:
      battotalenergydischarged:
      battotalenergydischarged:
      batdailyenergycharged:
      batdailyenergydischarged:
      griddailyongridenergy:
      griddailyenergypurchased:
      griddailyenergyused:
      gridpowergridtotalpower:
      gridtotalconsumptionpower:
and rebooted the VM

Configuration showed valid in the web UI.

Some new entities showed values but the date showed they were updated when I restarted the VM, and did not update.

I restarted the HA again via the WebUI - initially all solis entities showed 'unknown' - after a while (sorry, cannot say how long as I had to go out) most were populated but again they did not update, only showing the values from the restart time.

Also the Total Consumption still shows 'unknown', and the Total Grid Energy is 0

I have some screenshots if they might help. 
Is there something else I needed to do to install this update? or to test it?
TIA

hultenvp, 2021-11-01:
Hi @hawkeye100 

Thanks for testing.

At startup the integration schedules an update for one minute after startup. This is done to avoid load issues during startup.
So seeing "unknown" for one minute after a restart is as expected.
After the initial update the integration checks every 2 minutes if anything changed. If not, then also nothing is updated. Especially on a cloudy day it could take a lot of time before you see any changes.

That said there's still something wrong.

In issue #23 user Graham noted I made a typo in  Total power. 
```
(Typo on line 366 of "/config/custom_components/solis/platform2_portal.py" - GRID_DTOTAL_POWER should be GRID_TOTAL_POWER )
```
I fixed it in the meantime, but it might be you pulled your version before I fixed it.
If the bug is there then I expect the integration crashes and stops updating.

Could you check you have the latest code?
https://github.com/hultenvp/solis-sensor/tree/23-add-more-sensors-for-battery/custom_components/solis

If you do and the issue is still there then could you send me debug logging output? You can do that by adding the following lines to configuration.yaml and then restart the core.

```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```

Logging is found in Configuration->Logs->Load full home assistant log

Thanks,
Peter

hawkeye100, 2021-11-01:
I downloaded the individual files from the link you sent (individually right click - save link as)
The manifest.json file was quite different so I put the old one back (from my previous download of #23 files)

Rebooted the VM as configuration showed invalid from previous restart (solis did not exist)
This took some time.
Then most entities showed **Unavailable** (after more than 1 minute, not Unknown) - not just solis. (weather was OK) 

Checking configuration.yaml just circled  (did not wait to finish)
System would not restart from webUI error on line 7 of sensor.py

Debug on as suggested, log file attached - hope it supplies the relevant information
TIA

[errorlog1-ticket23-solis.txt](https://github.com/hultenvp/solis-sensor/files/7455162/errorlog1-ticket23-solis.txt)

hultenvp, 2021-11-01:
Hi @hawkeye100 ,

Looking at the log it complains about your sensor.py file.

It seems to contain HTML code, probably something went wrong while copying.
I've attached a zip file with all the right files.

Make sure at least the following files are copied to your solis directory:
/custom_components/solis/manifest.json
/custom_components/solis/const.py
/custom_components/solis/sensor.py
/custom_components/solis/platform2_portal.py

[solis-sensor-23-add-more-sensors-for-battery.zip](https://github.com/hultenvp/solis-sensor/files/7456136/solis-sensor-23-add-more-sensors-for-battery.zip)

hawkeye100, 2021-11-01:
Many thanks - obviously my method for downloading screwed the files up.
It appears to be working now and updating, though the following may need looking at:

- the icons mdi:transmission-tower-import and mdi:transmission-tower-export do not appear next to the entities: Dailyon grid energy and Daily energy purchased

- solar inverter energy this year shows as unavailable

I'll check on other aspects tomorrow once I have seen it running over a period of time.
Thanks again for your help sorting this out.

hultenvp, 2021-11-02:
Ah, I see what happened. I fixed a small typo in const.py: "energy this year" --> "Energy this year".
As a result HA thinks it is a different sensor and created sensor.solis_energy_this_year_2.

I've reverted the change and at least for me it works. I recommend you change the capital 'E' back in to a lower case 'e' in const.py (or you wait for the new release if you don't mind missing the data for now)

The icons I don't know. What version of Home Assistant core are you on?

hawkeye100, 2021-11-02:
Total energy this year is now back on stream (with lower case 'e' for now)
Icons - are not critical (for me) and can wait perhaps - my version of HA is 2021.10.6
[image]

The total grid power appears to be working OK (fortunately sunny here today) - positive and negative values showing which correlate with the soliscloud platform graphs
Total consumption also seems to reflect the relevant values on soliscloud (using same data as m.ginlong.com)

Many thanks for your work on this - should be really useful.

hultenvp, 2021-11-03:
Thanks for the update @hawkeye100 .

I've prepared the pull request and now waiting for the hacs  team to integrate this integration. Not sure what happens if I release a new version while waiting for hacs integration.
When hacs officially supports my integration I will prepare the new release.

The icons did also not work in the latest version, by the way. I changed them to MDI:transmission-tower. Better than no icon ;-)

hultenvp, 2021-11-09:
Merged and release in v0.3.3. No default repo in HACS yet

closed 2021-11-09 by hultenvp