hultenvp/solis-sensor #27: HACS cant find Solis repo/integration

deccos, 2021-11-02:
trying to add this integration via HACS, 2021.10 version of HA.  HACS cant find an integration called Solis  (or Ginlong)

hultenvp, 2021-11-02:
Hi @deccos ,

Thanks for your feedback.
You are correct. I've prepared the integration for HACS support, but waiting for HACS to include me in the next batch. I hope it will be there in a few days. That's why the readme already says it supports install via HACS
If you can't wait I propose you temporarily go via HACS->Integrations and then "custom repositories" in the hamburger menu.
Cheers

closed 2021-11-02 by hultenvp

Blokkert, 2021-11-12:
Hi, wondering about the current state for adding it to HACS? Few days has passed but i see no Solis yet :)

hultenvp, 2021-11-13:
I'm still waiting for integration. They seem to be adding batch-wise very irregularly. Not sure when it's my turn. 
You can follow issue #14 for progress. Once that one's closed then the integration is in HACS.

Cheers

Blokkert, 2021-11-13:
Awesome! No problem, did use the "custom repositorie" now, looks like its working! :)