hultenvp/solis-sensor #28: Multiple inverters

Friesje, 2021-11-07:
I'd like to add multiple solis inverters. How do i add them?

Both won't work if i add another solis sensor with a different name.

hultenvp, 2021-11-09:
Hi @Friesje 

I had one request before to support multiple instances, but that person never got back to me if the fix solved it.

I'm working on a better way, but it should work to configure 2 Solis instances.
First of all, please make sure you use the latest version (v0.3.3)

Then add a configuration like this:
```
# sensor SolisPV number 1:
  - platform: solis
    name: "Solis1"
    portal_domain: "m.ginlong.com"
    portal_username: "your user name" 
    portal_password: "your password"
    portal_plant_id: "plantid"
    inverter_serial: "serial"
    sensors:
      energytoday:
      energytotal:

# sensor SolisPV number 2:
  - platform: solis
    name: "Solis2"
    portal_domain: "m.ginlong.com"
    portal_username: "your user name" 
    portal_password: "your password"
    portal_plant_id: "plantid"
    inverter_serial: "serial"
    sensors:
      energytoday:
      energytotal:
```

That should give you the total energy for today and lifetime of both the inverters. It's not possible to provide a total for all inverters as a sensor, that's supposed to be added up in Home Assistant (for example energy dashboard).

If it does not work then I'm interested in the log output. Please add then the following lines in your configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```
and please attach that log output to this issue (do not forget to remove your personal info from the log).

Looking forward to your response.

Cheers.

Friesje, 2021-11-12:
This worked for me, thanks!

closed 2021-11-12 by Friesje

hultenvp, 2021-11-12:
Great to hear. Thanks for letting me know.