hultenvp/solis-sensor #29: Today view totalling previous day with today

salty2011, 2021-11-08:
when showing the energy view on home assistant this integration appears to be totalling the previous days production along with todays production, see image below

[image]

To be clear I have configured the solar panels to use the "Inverter Energy Today" sensor. Is this the correct sensor to be using or should I be using something else?

hultenvp, 2021-11-09:
Hi @salty2011 ,

Thanks for showing interest in my integration.
To be frank I have not seen this issue before.

You are using the right sensor, so that's not the problem.

Let's ensure it's not an energy dashboard issue.
I'd like to ask you to:
- Upgrade to core version 2021.11.x (I'm on 2021.11.1)
- Upgrade to the latest version of the Solis integration. I made a new release today (v0.3.3). Note: I'm still waiting to become a default HACs integration, for now use manual install or HACS custom repo option.

If that does not solve the issue then let's go for debugging. In that case please add the following lines to your configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```

and please send me the log output.

Looking forward to your response.

Cheers.

salty2011, 2021-11-10:
hi @hultenvp 

Weirdly, the issue is no longer occurring.. just seems to have been for that day.

But will performed the above recommendations anyway.

closed 2021-11-10 by salty2011