hultenvp/solis-sensor #31: Energy and power by PV string

joseap, 2021-11-18:
Hi,

I'll really appreciate the posibility to get the figures for energy and power by  PV string.
Also I want to confirm that this integration is working well with Sofar Solar inverters and  
if you have the credentials for access to home.solarman.cn, you can use the same in m.ginlong.com

Thanks for your work

hultenvp, 2021-11-19:
Thanks @joseap 

I can add power per string. Energy per string data is not provided by the backend.

joseap, 2021-11-19:
Hi,

 

Thank you for your quick response.

 

About energy, I know it. It was a mistake.

 

Regards

Enviado el: viernes, 19 de noviembre de 2021 9:15
Para: hultenvp/solis-sensor ***@***.***>
CC: joseap ***@***.***>; Mention ***@***.***>

hultenvp, 2021-11-20:
Hi @joseap ,

Just released 0.3.4 with DC string power support. Don't forget to add it in your configuration to actually see it in HA.
Have fun!

closed 2021-11-20 by hultenvp

joseap, 2021-11-20:
Thank you very much. I really appreciate it.

Enviado el: sábado, 20 de noviembre de 2021 17:37