hultenvp/solis-sensor #33: not showing power

lex53, 2021-11-27:
Hi

Ive set this up in HA but no data is coming through. Do I need the quote marks with the user name and password or is it with out quotes in the configuration.yaml file?

also do i need to enter login details elsewhere?

cheers
alex

hultenvp, 2021-11-27:
Hi Alex

Yes, domain, username, password, plant id, inverter serial are all in quote marks (")
Use the same credentials as you use to log in with m.ginlong.com

If that does not work please add the following to your configuration.yaml and please attach the log to this issue.

```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```

lex53, 2021-11-28:
Hi,

Thank you for getting back to me.
That seems to have worked, but I'm only getting the reading for the solar%
none of the other readings are coming through.

I'll get the debug logs over to you when I get a chance.

Thanks again. It's a really cool integration.

hultenvp, 2021-11-28:
If you only get one sensor then maybe your configuration contains a typo.
Make sure all sensors are written correctly and that you follow the YAML syntax, specifically indentation and semicolons.

Cheers
Peter

lex53, 2021-11-28:
Hi Ive pasted the code into the configuration.yaml file but silly question
how do I get the debug logs?
Kinda new to all this.

cheers

hultenvp, 2021-11-28:
In home assistant go to configuration, then logs. There select "show full home assistant log".

The log file is also stored in the same directory as the configuration.yaml file

Cheers

lex53, 2021-11-28:
hi, here are the logs.

Update of sensor.ecowater_ac000w006379439 is taking over 10 seconds
9:13:45 PM – (WARNING) helpers/entity.py
Error doing job: Task exception was never retrieved
8:44:31 PM – (ERROR) components/cloud/__init__.py
Connection problem to snitun server
8:43:37 PM – (ERROR) /usr/local/lib/python3.9/site-packages/hass_nabucasa/remote.py - message first occurred at 8:43:21 PM and shows up 2 times
Challenge/Response error with SniTun server (0 bytes read on a total of 32 expected bytes)
8:43:37 PM – (ERROR) /usr/local/lib/python3.9/site-packages/snitun/client/client_peer.py - message first occurred at 8:43:21 PM and shows up 2 times
Setup of binary_sensor platform mobile_app is taking over 10 seconds.
8:43:35 PM – (WARNING) Binary sensor
Setup of sensor platform ecowater_softener is taking over 10 seconds.
8:43:35 PM – (WARNING) Sensor - message first occurred at 8:43:31 PM and shows up 2 times
Config entry 'Lysander' for accuweather integration not ready yet; Retrying in background
8:43:34 PM – (WARNING) config_entries.py - message first occurred at 8:43:34 PM and shows up 2 times
 
2021-11-28 20:43:03 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration bhyve which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-11-28 20:43:03 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration ecowater_softener which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-11-28 20:43:03 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-11-28 20:43:03 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solis which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-11-28 20:43:03 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration xiaomi_cloud_map_extractor which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-11-28 20:43:17 DEBUG (MainThread) [custom_components.solis.sensor] Creating solis sensor: actualpower
2021-11-28 20:43:17 DEBUG (MainThread) [custom_components.solis.sensor] Creating solis sensor: energytoday
2021-11-28 20:43:17 DEBUG (MainThread) [custom_components.solis.sensor] Creating solis sensor: status
2021-11-28 20:43:17 DEBUG (MainThread) [custom_components.solis.sensor] Creating solis sensor: temperature
2021-11-28 20:43:17 DEBUG (MainThread) [custom_components.solis.sensor] Creating solis sensor: dcinputvoltagepv1
2021-11-28 20:43:17 DEBUG (MainThread) [custom_components.solis.sensor] Creating solis sensor: dcinputcurrentpv1
2021-11-28 20:43:17 DEBUG (MainThread) [custom_components.solis.sensor] Creating solis sensor: acoutputvoltage1
2021-11-28 20:43:17 DEBUG (MainThread) [custom_components.solis.sensor] Creating solis sensor: acoutputcurrent1
2021-11-28 20:43:17 DEBUG (MainThread) [custom_components.solis.sensor] Creating solis sensor: energylastmonth
2021-11-28 20:43:17 DEBUG (MainThread) [custom_components.solis.sensor] Creating solis sensor: energythismonth
2021-11-28 20:43:17 DEBUG (MainThread) [custom_components.solis.sensor] Creating solis sensor: energythisyear
2021-11-28 20:43:17 DEBUG (MainThread) [custom_components.solis.sensor] Creating solis sensor: energytotal
2021-11-28 20:43:17 DEBUG (MainThread) [custom_components.solis.sensor] Starting updates....
2021-11-28 20:43:17 DEBUG (MainThread) [custom_components.solis.platform2_portal] Scheduling next update in 1 minutes.
2021-11-28 20:43:21 ERROR (MainThread) [snitun.client.client_peer] Challenge/Response error with SniTun server (0 bytes read on a total of 32 expected bytes)
2021-11-28 20:43:21 ERROR (MainThread) [hass_nabucasa.remote] Connection problem to snitun server
2021-11-28 20:43:31 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform ecowater_softener is taking over 10 seconds.
2021-11-28 20:43:34 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Lysander' for accuweather integration not ready yet; Retrying in background
2021-11-28 20:43:34 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Philips hue' for hue integration not ready yet: Error connecting to the Hue bridge at 192.168.1.100; Retrying in background
2021-11-28 20:43:35 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform mobile_app is taking over 10 seconds.
2021-11-28 20:43:35 WARNING (MainThread) [homeassistant.components.binary_sensor] Setup of binary_sensor platform mobile_app is taking over 10 seconds.
2021-11-28 20:43:37 ERROR (MainThread) [snitun.client.client_peer] Challenge/Response error with SniTun server (0 bytes read on a total of 32 expected bytes)
2021-11-28 20:43:37 ERROR (MainThread) [hass_nabucasa.remote] Connection problem to snitun server
2021-11-28 20:44:19 INFO (MainThread) [custom_components.solis.platform2_portal] Login Successful!
[... cut ...]
2021-11-28 21:13:45 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.ecowater_ac000w006379439 is taking over 10 seconds
2021-11-28 21:14:46 DEBUG (MainThread) [custom_components.solis.platform2_portal] Scheduling next update in 2 minutes.
2021-11-28 21:16:50 DEBUG (MainThread) [custom_components.solis.platform2_portal] Scheduling next update in 2 minutes.
2021-11-28 21:18:52 DEBUG (MainThread) [custom_components.solis.platform2_portal] Scheduling next update in 2 minutes.
2021-11-28 21:20:53 DEBUG (MainThread) [custom_components.solis.platform2_portal] Scheduling next update in 2 minutes.

hultenvp, 2021-11-29:
The log looks clean (at least the Solis part). Sensors are registered and updates are scheduled.

I do not see why it should not be working.
Please check the following:
- Are you seeing the data you expect on the ginlong portal (m.ginlong.com)?
- Please double check plant_id and inverter_serial that they are 100% correct.

lex53, 2021-11-29:
Thanks for getting back

just to confirm I just add the script as per the instructions to
the /config/configuration.yaml file?

I dont need to add anything to /config/custom_components/solis/const.py? I
noticed there are entries for user name and PW etc.

Its definitely looking at the right inverter and Ive got the right ID in
there. I can see info like temp and voltage in m.ginlong.com so its odd its
only pulling the solar power only but not the others.

Cheers
Alex

Cheers
Alex

hultenvp, 2021-11-29:
Not sure that's what you are referring to with "script", but the logger: section must indeed be added to configuration.yaml

You do not need to modify const.py or any of the other source files.

Just to be sure: Did you install via HACS? Or did you use other means (e.g. HACS custom integration or manual install)?

lex53, 2021-11-29:
Hi

I just pasted this to the end of configuration.yaml in file editor. From
sensor: onwards

# Configure a default setup of Home Assistant (frontend, api, etc)
default_config:

# Text to speech
tts:
  - platform: google_translate

group: !include groups.yaml
automation: !include automations.yaml
script: !include scripts.yaml
scene: !include scenes.yaml

sensor:
  - platform: solis
    name: "My Solis Inverter"
    portal_domain: "m.ginlong.com"
    portal_username: ***@***.***"
    portal_password: "xxxxxxxxxxxxx"
    portal_plant_id: "xxxxxxxx"
    inverter_serial: "xxxxxxxxxxx"
    sensors:
      actualpower:
      energytoday:
      status:
      temperature:
      dcinputvoltagepv1:
      dcinputcurrentpv1:
      acoutputvoltage1:
      acoutputcurrent1:
      energylastmonth:
      energythismonth:
      energythisyear:
      energytotal:

logger:
  default: warning
  logs:
    custom_components.solis: debug

lex53, 2021-11-29:
Sorry, I installed it from HACS yes. I just hit the install button and then
added the script.

I followed this.

https://github.com/hultenvp/solis-sensor

hultenvp, 2021-11-29:
Odd, 

Looks like you only get the first sensor (actualpower).
Does it continue to update? I mean, would you see a graph if you let the integration run for a while?
What would happen if you make energytoday the first one, would you then only get energytoday?

Another thing I noticed is that (can't recall exact scenario) homeassistant decides that your sensor is a new one and creates for example a "energytoday1" sensor. Maybe check that you are actually receiving the data, but it is in a different sensor.

Peter

lex53, 2021-11-29:
HI Peter,

I think I found the issue. When I add an entity to a graph and search my
solis there are 2 values for everything and one of each is greyed out. The
other is in blue and that shows the correct values. Seems its made 2
entities for each but only one is working. so all good.

Thanks for your help and great work on the integration. I really like it.

Do you know if this will work on the lux power inverter?

Cheers
Alex

Does that make sense?

hultenvp, 2021-11-29:
Good to hear you managed to sort it out.
Good luck using the integration.

It looks Lux is a different technology. I do not see a monitoring portal like the ginlong one.

Closing the issue.
Cheers Peter

closed 2021-11-29 by hultenvp