hultenvp/solis-sensor #38: Cannot import name 'HTTP_OK'

benv666, 2021-12-05:
Running homeassistant docker - just pulled the latest beta, getting this error.
(note that I never had this integration working in previous versions, just added it to see if it would work)

```
Logger: homeassistant.config
Source: custom_components/solis/platform2_portal.py:21
Integration: solis (documentation, issues)
First occurred: 2:45:33 PM (1 occurrences)
Last logged: 2:45:33 PM

Platform error: sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config.py", line 887, in async_process_component_config
    platform = p_integration.get_platform(domain)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 530, in get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 535, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/custom_components/solis/sensor.py", line 44, in <module>
    from .platform2_portal import InverterData as inverter
  File "/config/custom_components/solis/platform2_portal.py", line 21, in <module>
    from homeassistant.const import HTTP_OK
ImportError: cannot import name 'HTTP_OK' from 'homeassistant.const' (/usr/src/homeassistant/homeassistant/const.py)

```

hultenvp, 2021-12-06:
Hi @benv666 ,

Can you please share your core version, how you run Home Assistant,  how you installed the integration (manual, HACS custom repo, HACS) and what version of my integration you are using?

This smells like an install issue so the more info you can share of your setup the better.

benv666, 2021-12-06:
Hi hultenvp,

As I said I run the docker variant of homeassistant, specifically `homeassistant/home-assistant:beta` which I pulled yesterday (after which I noticed these errors). The image has `"org.opencontainers.image.version": "2021.12.0b0"` as label.
I've used HACS for installation, which claims to be v.0.4.0.
In `custom_components/platform2_portal.py` I can indeed see the import on line 21 `from homeassistant.const import HTTP_OK`

Some googling lead me to https://github.com/home-assistant/core/commit/fbb7c3136847a42339965614013e7a7014fb946b which shows the HTTP_OK has been removed, so I take it you'll need to work around that.
Let me know if there's anything else I can provide, and thanks for the support! :)

hultenvp, 2021-12-06:
Hi Ben,

Thanks, overlooked your first sentence.
That's a good catch. Some further digging in the commits of the same author gives me this link: https://github.com/home-assistant/core/commit/b0e1bab58b6c4d662a4b3ed88554180db645e1fa

Guess that gives me a solution direction. ;-)
Thanks again for spotting

hultenvp, 2021-12-06:
v0.4.1 created. Should be available in a while.
Please let me know if this fixes the issue for you.

benv666, 2021-12-07:
Yup, now loads without a problem.
Thanks for the support!

closed 2021-12-07 by benv666