hultenvp/solis-sensor #41: Error in custom component

Sparky699, 2021-12-15:
Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solis/platform2_portal.py", line 218, in async_update
    result = await self.interface_portal.async_update()
  File "/config/custom_components/solis/platform2_portal.py", line 616, in async_update
    await self.update_device_id()
  File "/config/custom_components/solis/platform2_portal.py", line 516, in update_device_id
    for record in resultJson['result']['paginationAjax']['data']:
TypeError: string indices must be integers

HomeAssistant Version information:
core-2021.12.1
Python-version | 3.9.7
Operatingsystem | Linux
Version | 5.10.63-v8

hultenvp, 2021-12-15:
Hi @Sparky699 : Not the best error, but this occurs when your PlantId is incorrect.
Note that plantId must be a string in your configuration.yaml, see manual

Sparky699, 2021-12-16:
Hi! Thanks that did the trick. I have two accounts and the plantID was from the other login. If you can trigger a message when the PlantID isn't recognized it's easier to notice. But thank you so much... have been looking for a long time to get my inverters going! Now the fun stuf can begin. Trigger the washing machine when there is enough sun!
Thnx!

closed 2021-12-16 by Sparky699