hultenvp/solis-sensor #42: Refresh rate

radzik80, 2021-12-15:
Hello, 

I have some problems with updating data from ginlong. As I found in code there should be refresing every two minutes. In my case there is only updated at HA restart.

Any suggestions?

Regards

hultenvp, 2021-12-15:
Hi @radzik80: The code compares the new value to the previous and only notifies HomeAssistant of a change if there is any.
In night and on dark days updates could be very infrequent.

If there is sun and you still do not see updates then add the following to your configuration.yaml and attach the log to this issue 

```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```

radzik80, 2021-12-15:
Ok I will check at daytime.
Thank you for quick response.

radzik80, 2021-12-16:
> 

Hello, it looks like you described.
Thank you for your support @hultenvp 

> Hi @radzik80: The code compares the new value to the previous and only notifies HomeAssistant of a change if there is any. In night and on dark days updates could be very infrequent.
>
> If there is [...]

closed 2021-12-16 by radzik80