hultenvp/solis-sensor #43: Daily incorrect readin

peterdusink, 2021-12-17:
Hi,

Almost daily there;s an incorrect reading from sensor.my_solis_inverter_energy_today. It looks like the production of previous day is shown only between 07 and 08 am. 
[image]

Not sure why this happens and how to fix.

Not the energy dashboard is unusable as it picks up this reading :-(

Actual production today is 0.2 KWH

[image]

Energy dashboard shows 

[image]

hultenvp, 2021-12-19:
Hi @peterdusink,

There is a fix in the code that resets the energy today at midnight. Without this fix it would happen at sunrise and that would skew daily figures. 
My guess is that for some reason the value reverts until the values of the new day come in. 
I'll need more days to figure out what triggers the reversal.

Could you share with me the exact values before reset and at the red arrow?

Thanks

peterdusink, 2021-12-20:
Hi @hultenvp,

Thank you for the reply. 

The values before reset and red arrow are the same. 

But I think to know what triggers the reversal. 

I have been editing dashboards and configuration in the last days. It seem to only happen when I'm editing configuration.yaml and performing some restarts to apply the updated configuration.yaml

I didn't see value reverting by restart HA only (without modified configuration.yaml)

Today I did not make any changes, or restart. Now the solar readings are correct :-)

Thanks.

hultenvp, 2021-12-21:
Ah, that indeed will cause the integration to revert.
Looks like you need to postpone your work on HomeAssistant to later on the day in winter time ;-)

Thanks for getting back, closing issue.

closed 2021-12-21 by hultenvp