hultenvp/solis-sensor #45: Integrate New Energy Constants into solis-sensor

conallob, 2022-01-04:
I installed this sensor via HACS and configured it with the new Energy Management features introduced since August 2021.

Some sensors integrate cleanly while other sensors (e.g power supplied to a grid, battery charging and discharging, etc) have wildly in accurate values.

A cursory look at both https://github.com/hultenvp/solis-sensor/blob/master/custom_components/solis/const.py and https://github.com/home-assistant/core/blob/dev/homeassistant/components/energy/sensor.py tells me solis-sensor is using custom defined consts which predate the consts now defined in home-assistant/core/

In order to facilitate clean integration with the energy management functionality, https://github.com/hultenvp/solis-sensor/blob/master/custom_components/solis/const.py should utilise the new consts in home-assistant/core if the core version is suitably recent.

hultenvp, 2022-01-04:
Hi @conallob, thanks for the feedback

I've seen those constants and am taking this along in a code cleanup I'm doing at the moment.

Do you have any insights why not using these constants would decrease the accuracy of the measurements? While I agree the integration should use predefined constants from core I do not see how it affects accuracy.

For battery there is a difference in precision between the various sensors (see: https://github.com/hultenvp/solis-sensor#readme), would that at least for battery be the source of your issues?

conallob, 2022-01-04:
My bad, I did not provide you with enough context.

When configured, the Energy Management tab reveals my household demand is 65525 kWh and my battery charge capacity is 131072 kWh, which are both definitely off. (See https://photos.app.goo.gl/VAn8ra7UjBPzmSCT7 for a screenshot)

65525 is suspiciously close to 65536 and 131072 is 2x 65536, which made me think max int is being misreported. This hunch led me to looking at the consts being used.

hultenvp, 2022-01-04:
That is indeed fishy.

Do you see the same values in the grid and battery sensor history?

hultenvp, 2022-01-16:
Hi @conallob,

You can find a test version that supports the energy constants here:
https://github.com/hultenvp/solis-sensor/tree/48_Rewrite_api_code_and_introduce_mypy_supported_typing

Interested in your feedback if it fixes your issue.

closed 2022-01-20 by hultenvp

conallob, 2022-01-20:
Thank you for the quick turnaround. I've been running this version for ~24 hours now without any major issues.

The Developer Console does still report "The unit ('kWh') of this entity doesn't match a unit of device class 'power'." for 
sensor.solis_inverter_total_energy_used however. It's current value is 51,642,438.0 kWh