hultenvp/solis-sensor #46: When inverter is turned on kwh from yesterday is reported

henrikeri, 2022-01-11:
Whenever the inverters start generating power, the kwh production from last power on is reported and deducted. This messes up the statistics and energy dashboard. See attached screenshots from m.ginlong.com (not showing it) and HA instances showing the report.

Is this something that needs solving in the sensor, or on my side?

From HA:
[image]
[image]
From m.ginlong.com
[image]

hultenvp, 2022-01-12:
Hi @henrikeri ,

There is a patch in the sensor implementation to avoid this from happening. 
Energy today is only reset to 0 in the morning when the inverter switches on again. That means that for the whole nighttime the backend service reports the energy produced on the past day. 
Without fix this would indeed mess up the energy dashboard. 
So I now report back the value of 0 for as long as the energy of today is exactly the energy of yesterday.
This works like a charm unless you restart home assistant between midnight and startup of the inverter. Then energy_yesterday value is lost and the sensor can no longer compare and returns whatever it gets from the backend (which is the old data) 
I suspect that that is what happens in your case. Can you confirm?

I'm looking into more robust solutions that survive a restart, but that's still work in progress as I did not yet find a reliable datasource.

henrikeri, 2022-01-12:
Hi @hultenvp,
I will verify when the system comes back online again. Sun is still a bit late here in Norway. As for surviving reboots, could we possibly write and read the variable from a file stored in custom_components? Should survive reboots, right?

hultenvp, 2022-01-12:
Not sure if writing files there is allowed, regardless I prefer a stateless solution. I'm currently following another lead where I use additional backend data to reliably establish whether the value I'm seeing is actually from today or still from yesterday. Think it will work.
I'm in the last stages of a rewrite of the code. It will be part of that change. When done I'll inform you.

hultenvp, 2022-01-16:
Hi @henrikeri,

You can find a test version that fixes the issue here:
https://github.com/hultenvp/solis-sensor/tree/48_Rewrite_api_code_and_introduce_mypy_supported_typing

Interested in your feedback

henrikeri, 2022-01-17:
hI @hultenvp,

I can confirm that the issues were indeed related to restarting home assistant after sunset (which is early in Norway atm). I have just installed the test version and will keep you posted on any issues that might appear!

Best regards,
Henrik

closed 2022-01-20 by hultenvp