hultenvp/solis-sensor #47: New extension request for yesterday energy sensor

jonczykp, 2022-01-11:
First, I would like to appreciate the work you did.

Is there a possibility to to provide one more sensor energyyeserday?. My energy supplier provides the statistic with one day delay so energy dashboard does not show the proper data. For sure I will be not able to get online data, but yesterday is also OK for me.

I tired to go trough the code to review JSONs from API but getting strange error when login like: HTTPSConnectionPool(host='m.ginlong.com', port=443): Max retries exceeded with url: /cpro/login/validateLogin.json?userName=......&password=..........&lan=2&domain=m.ginlong.com&userType=C (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:1108)')))

hultenvp, 2022-01-12:
Hi @jonczykp ,

Energy yesterday is not a value that is returned by  the ginlong backend. I cannot return that info in a reliable way as the information is lost in a restart.
I propose you see if you can make a SQLsensor that gets that data from the solis sensor for your use case.
https://www.home-assistant.io/integrations/sql/

closed 2022-01-12 by hultenvp

hultenvp, 2022-01-12:
Or maybe utility meter can be put to use for your case:
https://www.home-assistant.io/integrations/utility_meter

It is persistent

jonczykp, 2022-01-12:
Hi @hultenvp,
Thank for suggestions, it seems it can be the only solution.
But maybe you know the URL for JSON that takes detailed data of production,, we see this data on the first page in the portal. I could try to develop some code for this.