hultenvp/solis-sensor #52: Sensor unable to connect since update to 1.0.0

salty2011, 2022-01-21:
Hello

Since updating from 0.4.1 to 1.0.0 the services does not see to be able to retrieve data. 

Within the log files I am seeing the following error

`Logger: homeassistant.components.sensor
Source: custom_components/solis/ginlong_api.py:340 
Integration: Sensor (documentation, issues) 
First occurred: 5:40:29 PM (1 occurrences) 
Last logged: 5:40:29 PM

Error while setting up solis platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solis/sensor.py", line 162, in async_setup_platform
    sensors = await async_autodetect(service)
  File "/config/custom_components/solis/sensor.py", line 84, in async_autodetect
    capabilities = await inverter_service.discover()
  File "/config/custom_components/solis/service.py", line 86, in discover
    data = await self._api.fetch_inverter_data(inverter_serial)
  File "/config/custom_components/solis/ginlong_api.py", line 294, in fetch_inverter_data
    self._post_process()
  File "/config/custom_components/solis/ginlong_api.py", line 340, in _post_process
    self._data[INVERTER_TIMESTAMP_ONLINE] = float(self._data[INVERTER_TIMESTAMP_ONLINE])/1000
KeyError: 'timestampOnline'`

also get

`Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/connection.py:89 
Integration: Home Assistant WebSocket API (documentation, issues) 
First occurred: 5:40:25 PM (1 occurrences) 
Last logged: 5:40:25 PM

[140113332918256] Received invalid command: energy/info`

I have also updated my configuration to match the updates in the readme and look like below (obscured my details for privacy / security reasons)

`sensor:
  - platform: solis
    name: "My Solis Inverter"
    portal_domain: "m.ginlong.com"
    portal_username: !secret solis_username
    portal_password: !secret solis_password
    portal_plant_id: "00000"`

I am running HA 2021.12.10, is there something I am missing to make this working. Can confirm was working on version 0.4.1

hultenvp, 2022-01-21:
Cannot reproduce:
Version: 2021.12.10
OS: 7.1
Solis: v1.0.0

Can you unquote your plantID, so it is an integer instead of string?

Cheers

hultenvp, 2022-01-24:
Hi @salty2011, do you still have issues on v1.0.1?

salty2011, 2022-01-25:
Hi @hultenvp 

Can confirm that plantid is not in "" and still getting issue on version 1.0.1

Added the logging sensor as mentioned above.

In the logs I am getting the following errors reported

`Logger: homeassistant.components.sensor
Source: custom_components/solis/ginlong_api.py:338
Integration: Sensor (documentation, issues)
First occurred: 5:38:39 PM (1 occurrences)
Last logged: 5:38:39 PM

Error while setting up solis platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solis/sensor.py", line 162, in async_setup_platform
    sensors = await async_autodetect(service)
  File "/config/custom_components/solis/sensor.py", line 84, in async_autodetect
    capabilities = await inverter_service.discover()
  File "/config/custom_components/solis/service.py", line 92, in discover
    capabilities = await self._do_discover()
  File "/config/custom_components/solis/service.py", line 111, in _do_discover
    data = await self._api.fetch_inverter_data(inverter_serial)
  File "/config/custom_components/solis/ginlong_api.py", line 292, in fetch_inverter_data
    self._post_process()
  File "/config/custom_components/solis/ginlong_api.py", line 338, in _post_process
    self._data[INVERTER_TIMESTAMP_ONLINE] = float(self._data[INVERTER_TIMESTAMP_ONLINE])/1000
KeyError: 'timestampOnline'
`
Under my home assistant log file i do find the following

`2022-01-25 17:38:38 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-01-25 17:38:39 INFO (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 00000000
2022-01-25 17:38:39 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up solis platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solis/sensor.py", line 162, in async_setup_platform
    sensors = await async_autodetect(service)
  File "/config/custom_components/solis/sensor.py", line 84, in async_autodetect
    capabilities = await inverter_service.discover()
  File "/config/custom_components/solis/service.py", line 92, in discover
    capabilities = await self._do_discover()
  File "/config/custom_components/solis/service.py", line 111, in _do_discover
    data = await self._api.fetch_inverter_data(inverter_serial)
  File "/config/custom_components/solis/ginlong_api.py", line 292, in fetch_inverter_data
    self._post_process()
  File "/config/custom_components/solis/ginlong_api.py", line 338, in _post_process
    self._data[INVERTER_TIMESTAMP_ONLINE] = float(self._data[INVERTER_TIMESTAMP_ONLINE])/1000`

Seems to be authenticating with out issue. Problem occurs when it tries to setup the platform sensor. For testing I also tried change the name in my configuration to see if this would resolve the issue, but same issue.

salty2011, 2022-01-25:
just a note, in the above i changed the value for "Fetching data for serial: 00000000" when pasting here as its a public forum. In the actual log correct serial is shown

hultenvp, 2022-01-25:
Thanks for clarifying, was already wondering why serial '000000' was returned ;-)

I do see what's going wrong, I just don't get why you are not receiving the timestamp.

The keyerror is a robustness issue anyway, so I created a new patch version that at least does not trip over the missing key. It should be available in HACS when HACS picks it up in an hour or so. Not sure if it fixes the issue. If it is not helping then please remove the comment on line 290 in ginlong_api.py . It will dump the received server response in the log. Please attach that to the issue (remove sensitive details). It helps me determine what's different in your response compared to that of others.

hultenvp, 2022-01-25:
Please wait for v1.0.3, v1.0.2 is broken.

zoezoevp, 2022-01-25:
Hi,

The integration worked fine for until I upgraded to 1.0.2.  I saw 1.0.3 appearing quickly afterwards but it is still broken.  I have tried to roll back to a version prior 1.0.2 but it doesn't work anymore.  It stays broken.
This is what I see in the logs for 1.0.3 :

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solis/sensor.py", line 162, in async_setup_platform
    sensors = await async_autodetect(service)
  File "/config/custom_components/solis/sensor.py", line 84, in async_autodetect
    capabilities = await inverter_service.discover()
  File "/config/custom_components/solis/service.py", line 92, in discover
    capabilities = await self._do_discover()
  File "/config/custom_components/solis/service.py", line 111, in _do_discover
    data = await self._api.fetch_inverter_data(inverter_serial)
  File "/config/custom_components/solis/ginlong_api.py", line 292, in fetch_inverter_data
    self._post_process()
  File "/config/custom_components/solis/ginlong_api.py", line 338, in _post_process
    if self._data.has_key(INVERTER_TIMESTAMP_ONLINE):
AttributeError: 'dict' object has no attribute 'has_key'

salty2011, 2022-01-25:
@hultenvp just installed 1.0.3 and still getting error and looks like it still not loading as a result. Uncommented line 290 and have the below information

Logger: homeassistant.components.sensor
Source: custom_components/solis/ginlong_api.py:338
Integration: Sensor (documentation, issues)
First occurred: 10:27:21 PM (1 occurrences)
Last logged: 10:27:21 PM

Error while setting up solis platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solis/sensor.py", line 162, in async_setup_platform
    sensors = await async_autodetect(service)
  File "/config/custom_components/solis/sensor.py", line 84, in async_autodetect
    capabilities = await inverter_service.discover()
  File "/config/custom_components/solis/service.py", line 92, in discover
    capabilities = await self._do_discover()
  File "/config/custom_components/solis/service.py", line 111, in _do_discover
    data = await self._api.fetch_inverter_data(inverter_serial)
  File "/config/custom_components/solis/ginlong_api.py", line 292, in fetch_inverter_data
    self._post_process()
  File "/config/custom_components/solis/ginlong_api.py", line 338, in _post_process
    if self._data.has_key(INVERTER_TIMESTAMP_ONLINE):
AttributeError: 'dict' object has no attribute 'has_key'

not super strong in python but looks like on like 338 you have used has_key, according to some research it would appear that this was removed in python3.... I changed this to __contains and this appears to be working for me.

Will do a pr with the changes

salty2011, 2022-01-25:
Tested by modifying the files as mentioned above and appears to be working correctly now. Have submitted PR with updates

https://github.com/hultenvp/solis-sensor/pull/61/commits/890a25bc589ac1ce671c916a7f36dc27247c7450

zoezoevp, 2022-01-25:
> Tested by modifying the files as mentioned above and appears to be working correctly now. Have submitted PR with updates
> 
> [890a25b](https://github.com/hultenvp/solis-sensor/commit/890a25bc589ac1ce671c916a7f36dc27247c7450)

I also did the modification on my side and I can confirm it also fixed it for me.  thx

hultenvp, 2022-01-25:
Thanks all for confirming and indeed the has_key does no longer work. Tried to provide a quick fix during coffee break without proper testing. Me bad :flushed:

@salty2011: Do I understand it correctly that the actual issue is also solved? Or are you still experiencing issues?

salty2011, 2022-01-25:
@hultenvp correct the issue appear to be solved in my testing. Debugging is also showing data that it pulled from the initial query on load. 

Have had change in for about  an hours still appears to be working without problem.

The PR I submitted is to add the required changes. Apologies if not submitted entirely correct its my first one so was trying to follow what had been done previously.