hultenvp/solis-sensor #53: After upgrading to version 1.0.0 no solis sensors are being published

soheiladeli, 2022-01-22:
Hi @hultenvp 

After upgrading to version 1.0.0, now no solis sensors are being published. It was working on the previous version 0.4.1 where I had the sensors manually added in configuration yaml.

So what I did, I removed existing version via hacs, restarted HA, removed all existing sensors, installed new version 1.0.0 via hacs, added the configs to configuration.yaml, restarted HA, but there are no solis sensors added.

here is my configuration.yaml (jus put dummy username and password  and plant ID here and made sure the plant id is integer):

sensor:
  - platform: solis
    name: "Solis Inverter"
    portal_domain: "m.ginlong.com"
    portal_username: "username"
    portal_password: "passwoed"
    portal_plant_id: 12345678

HA version : core-2021.12.4
python_version | 3.9.9

here is the full logs after a restart: (obscured  the serial number)

2022-01-22 21:10:54 INFO (SyncWorker_1) [homeassistant.loader] Loaded solis from custom_components.solis
2022-01-22 21:10:54 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration solis which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-01-22 21:11:31 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform solis is taking over 10 seconds.
2022-01-22 21:11:33 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error:
2022-01-22 21:11:33 DEBUG (MainThread) [custom_components.solis.sensor] {}
2022-01-22 21:11:33 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes.
2022-01-22 21:12:59 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error:
2022-01-22 21:12:59 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes.
2022-01-22 21:14:03 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-01-22 21:14:05 INFO (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 0308E2211XXXXXX
2022-01-22 21:14:06 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 2 minutes.
2022-01-22 21:16:06 INFO (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 0308E2211XXXXXX
2022-01-22 21:16:09 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 2 minutes.

hultenvp, 2022-01-22:
Hi @soheiladeli 

plant_id shouldn't be a problem, made the code robust for int and str. Bit strange that you get an unspecified error logged, something's going wrong in communication with the server, resulting in empty responses. Login seems fine, which makes me  believe username and password are at least OK.
Can you double-check the plant-id?
I'll check how that unspecified error is produced.

Cheers

hultenvp, 2022-01-22:
Could you try commenting out line 292 in ginlong_api.py and post the result?
It will put the server response in the debug log.

soheiladeli, 2022-01-22:
Thanks @hultenvp. double checked the plant ID again and confirm it is correct.

also just checked the line 292 in the ginlong_api.py and it is already commented out,

[image]

hultenvp, 2022-01-22:
Sorry, I meant uncommenting..
I'd like to see the payload coming from the server, this line just dumps raw payload data in the log.

The log you provided shows that initially communication fails, but the integration appears to recover and continue.
My current guess is that communication fails during autodiscover and that no sensors get registered. After that the communication seems to go fine, but since no sensors got registered there's nothing to update.
The log could confirm.

soheiladeli, 2022-01-22:
ok I did, this the log:

2022-01-22 23:08:59 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error:
2022-01-22 23:08:59 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes.
022-01-22 23:10:00 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-01-22 23:10:00 INFO (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 0308E2211030003
2022-01-22 23:10:02 DEBUG (MainThread) [custom_components.solis.ginlong_api] {'result': {'datalogger': {'address': 'AustraliaNew South WalesHornsby ShireCherrybrookCounty Drive115 County Dr, Cherrybrook NSW 2126, Australia', 'datalogger': {'commandType': '02', 'gatewayModel': 'MW-3238071', 'sn': '4037610985', 'type': 3, 'updateDate': 1642810630000}, 'dataloggerSn': '4037610985', 'dataloggerType': '3', 'extend': '{"_st":"1","_et":"1642833712","a":"02","stat":"0","b":"69977","c":"19","d":"0","e":"5","f":"60","za":"4037610985","g":"1","h":"MW_08_512_0501_1.82","zb":"A5560010410001E90DA9F002591101001300000000000000053C78014A034D575F30385F3531325F303530315F312E383200000000000000000000000000000000000000000034EAE74F06F23139322E3136382E302E31383200000000000101055015","zc":"60.227.43.32:18066","zd":"1642810630","ze":"1642810631","zf":"","l":"74","zg":"00,01","m":"0","n":"1","zh":"47.88.8.200:10000","zi":"01","o":"0501","ruleCode":"010102","p":"3","zk":"1","zl":"0","t":"120","x":"34EAE74F06F2","y":"192.168.0.182","_ut":"1642810656"}', 'firmwareVersion': '无', 'hardwareVersion': '无', 'isDel': 0, 'lat': -33.7254575, 'lon': 151.0369696, 'plantId': 1510534, 'plantName': 'Adeli Solar Plant', 'productModel': 'MW-3238071', 'state': 2, 'updateDate': 1642810630000}, 'deviceDisplayName': 'Inverter', 'lngLat': '{"add":"AustraliaNew South WalesHornsby ShireCherrybrookCounty Drive115 County Dr, Cherrybrook NSW 2126, Australia","lng":151.0369696,"name":"Inverter","href":"http://pro.solarman.cn/index/setIndex.do?type=24&param1=1510534","lat":-33.7254575,"plantName":"Adeli Solar Plant","status":2}', 'deviceWapper': {'address': 'AustraliaNew South WalesHornsby ShireCherrybrookCounty Drive115 County Dr, Cherrybrook NSW 2126, Australia', 'data': '{"_st":"1","dt":"1642830111000","1ab":"1921","1af":"240.90","1ag":"239.70","1ah":"241.30","1ai":"2.60","1aj":"2.60","1ak":"2.60","1ao":"1840","1ar":"49.97","1ez":"1022","za":"4037610985","1aw":"21.90","ah":"311","zc":"60.227.43.32:18066","zd":1642830111000,"ze":1642830112000,"zf":"104505729","zg":"87,E9","zh":"47.88.8.200:10000","zi":"02","zk":"1","zl":"0","1bc":"50.00","1bd":"24.70","1be":"50.8","1ru":"0","1bf":"50.8","1rw":"4","1jq":"22-1-22 16:41:29","1fs":"","_ut":"1642830130","1bs":"1840","1bt":"1.00","1sm":"0026","bp":"1","_et":"1642833712","bq":"3","1ki":"00F4","br":"3","1a":"260.50","1b":"405.00","1j":"2.40","1k":"3.20","1gr":"8e","13ah":"46.10","1gt":"1","_":"1","a":"01","stat":"0","b":"0507","c":"89608","d":"19515","e":"1642740503","f":"0001","g":"0308E2211030003","1lf":"105","a0":"1","ruleCode":"0102010705","1df":"46.10","1hk":"6000"}', 'dataJSON': {'_st': '1', 'dt': '1642830111000', '1ab': '1921', '1af': '240.90', '1ag': '239.70', '1ah': '241.30', '1ai': '2.60', '1aj': '2.60', '1ak': '2.60', '1ao': '1840', '1ar': '49.97', '1ez': '1022', 'za': '4037610985', '1aw': '21.90', 'ah': '311', 'zc': '60.227.43.32:18066', 'zd': 1642830111000, 'ze': 1642830112000, 'zf': '104505729', 'zg': '87,E9', 'zh': '47.88.8.200:10000', 'zi': '02', 'zk': '1', 'zl': '0', '1bc': '50.00', '1bd': '24.70', '1be': '50.8', '1ru': '0', '1bf': '50.8', '1rw': '4', '1jq': '22-1-22 16:41:29', '1fs': '', '_ut': '1642830130', '1bs': '1840', '1bt': '1.00', '1sm': '0026', 'bp': '1', '_et': '1642833712', 'bq': '3', '1ki': '00F4', 'br': '3', '1a': '260.50', '1b': '405.00', '1j': '2.40', '1k': '3.20', '1gr': '8e', '13ah': '46.10', '1gt': '1', '_': '1', 'a': '01', 'stat': '0', 'b': '0507', 'c': '89608', 'd': '19515', 'e': '1642740503', 'f': '0001', 'g': '0308E2211030003', '1lf': '105', 'a0': '1', 'ruleCode': '0102010705', '1df': '46.10', '1hk': '6000'}, 'dataloggerSn': '4037610985', 'deviceBasicInfo': [{'key': 'g', 'name': 'SN', 'value': '0308E2211030003'}, {'key': '1gr', 'name': 'Product type', 'value': '8e'}, {'key': '1gt', 'name': 'single / three-phase type', 'value': '1'}, {'key': '1lf', 'name': 'Safety country', 'value': '105'}, {'key': '1rw', 'name': 'Power curve version', 'value': '4'}, {'baseUnit': 'W', 'key': '1hk', 'name': 'Rated power', 'unit': 'kW', 'value': '6.0'}], 'deviceId': 104505729, 'deviceType': 1, 'deviceVersionInfo': [{'key': '1sm', 'name': 'HMI version', 'value': '0026'}, {'key': '1ki', 'name': 'DSP version', 'value': '00F4'}], 'extend': '{"_st":"1","_et":"1642833712","a":"02","stat":"0","b":"69977","c":"19","d":"0","e":"5","f":"60","za":"4037610985","g":"1","h":"MW_08_512_0501_1.82","zb":"A5560010410001E90DA9F002591101001300000000000000053C78014A034D575F30385F3531325F303530315F312E383200000000000000000000000000000000000000000034EAE74F06F23139322E3136382E302E31383200000000000101055015","zc":"60.227.43.32:18066","zd":"1642810630","ze":"1642810631","zf":"","l":"74","zg":"00,01","m":"0","n":"1","zh":"47.88.8.200:10000","zi":"01","o":"0501","ruleCode":"010102","p":"3","zk":"1","zl":"0","t":"120","x":"34EAE74F06F2","y":"192.168.0.182","_ut":"1642810656"}', 'isDel': 0, 'lat': -33.7254575, 'lon': 151.0369696, 'minllis': 660, 'paramAllSelectors': '[{"key":"1bf","name":"Annual Generation (Active)","unit":"kWh","value":"50.8"}]', 'paramDaySelectors': '[{"baseUnit":"V","key":"1a","n [... cut ...] / three-phase type', 'value': '1'}, {'key': '1lf', 'name': 'Safety country', 'value': '105'}, {'key': '1rw', 'name': 'Power curve version', 'value': '4'}, {'baseUnit': 'W', 'key': '1hk', 'name': 'Rated power', 'unit': 'kW', 'value': '6.0'}], 'deviceVersionInfo': [{'key': '1sm', 'name': 'HMI version', 'value': '0026'}, {'key': '1ki', 'name': 'DSP version', 'value': '00F4'}]}}, 'state': 1}
2022-01-22 23:12:03 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 2 minutes.

hultenvp, 2022-01-22:
Thanks. I'll have a look. 

I've removed your log from the issue and made a copy as it contained sensitive data. I'll delete the log when I'm done analyzing.

Cheers

hultenvp, 2022-01-22:
Hi @soheiladeli, 

I've created a patch that should fix your issues. It will retry 3x during discover before giving up and also be more explicit in log when giving up.
You can find the code here:
https://github.com/hultenvp/solis-sensor/archive/refs/heads/53_54_robustness_issues.zip

Just overwrite your existing code with the files from the zip and restart.
If it fixes your issue then I'll release an official patch release. You can then resume updating normally via HACS.

soheiladeli, 2022-01-23:
Thanks @hultenvp 
Ok, I replaced the files and it actually worked, after two failed discovery on the third attempt and created all the sensors.

here is the log:

2022-01-23 11:20:33 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform solis is taking over 10 seconds.
2022-01-23 11:20:33 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error:
2022-01-23 11:20:33 INFO (MainThread) [custom_components.solis.service] Discovery failed, retry #0
2022-01-23 11:21:11 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error:
2022-01-23 11:21:11 INFO (MainThread) [custom_components.solis.service] Discovery failed, retry #1
2022-01-23 11:21:37 ERROR (MainThread) [homeassistant.components.sensor] Setup of platform solis is taking longer than 60 seconds. Startup will proceed without waiting any longer.
2022-01-23 11:21:51 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error:
2022-01-23 11:21:51 INFO (MainThread) [custom_components.solis.service] Discovery failed, retry #2
2022-01-23 11:22:06 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-01-23 11:22:08 INFO (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 0308E2211XXXXXX
2022-01-23 11:22:09 DEBUG (MainThread) [custom_components.solis.sensor] Creating inverterstate (0308E2211XXXXXX)
2022-01-23 11:22:09 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute state for inverter 0308E2211XXXXXX
.
.
.
2022-01-23 11:22:09 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute energyTotalLife for inverter 0308E2211XXXXXX
2022-01-23 11:22:09 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes.

soheiladeli, 2022-01-23:
Hi @hultenvp, Just FYI, one other issue I noticed while checking the sensors,

The "Solis Inverter AC Output Total Power" is displaying the unit as W, but I think it should be kW, below comparing home assistant sensor with the Solis online platform:

[image]

[image]

hultenvp, 2022-01-23:
Thanks for testing, I'll push the fix to release then and close this issue

Regarding the AC Output total power: That's worth a new ticket. Looks like the unit varies with power. For me it's in Watt's, but that's probably because I'm looking at a fully overcast winter sky and don't produce over 1kWh.

I'll create the ticket.

PR #55 (2022-01-23): 53 54 robustness issues
Fixes #53 
Fixes #54
closed 2022-01-23 by hultenvp