hultenvp/solis-sensor #54: V1.0.0 not robust enough in avoiding energy today reporting yesterday's energy.

hultenvp, 2022-01-22:
Still seeing spikes at startup, messing up energy dashboard.
Looks like energy today is not always reset already when state transitions to 'ON'.

Add grace period of ~10 minutes after inverter switching on before accepting 'energy today' values.

[image]

henrikeri, 2022-01-28:
@hultenvp 
Not sure if the grace period is long enough, I had yesterdays energy populate this morning. Should we re-open? 
[image]

hultenvp, 2022-01-28:
The grace period is not the problem. The problem is that the dir method sorts the list of attributes I return. Apparently that's acoording to spec, but it means 'energy_today' gets evaluated before 'status' and thus my check has no value as the timestamp is wrong. 
I'm testing an update, it worked nice this morning.  Plan to make a new release when also the temperature issue gets a conclusion.

Reopening this issue to get it included in release notes
[image]

PR #55 (2022-01-23): 53 54 robustness issues
Fixes #53 
Fixes #54
PR #72 (2022-01-28): 54 energy today robustness
Closes #54
closed 2022-01-28 by hultenvp