hultenvp/solis-sensor #56: AC Output power has variable unit of power

hultenvp, 2022-01-23:
Unit of power for AC output power returned by API (and possibly all power units?) can change between 'W' and 'kW'.
Sensor has a static unit, so proposal is to convert kW back to W

flytomcio, 2022-01-27:
I have the same issue...

suneworld, 2022-01-27:
Is it as simple that kW is output with a punctuation delimiter and W is not? If yes it is easy to fix with a HA template sensor.

hultenvp, 2022-01-27:
No it is that the unit of the data retrieved from the portal changes. The integration assumes it is in Watt, so if you are in a particular sunny location on this earth (not here!) then the portal returns for example 2.1kW, but it gets interpreted as 2.1W.
I need to also read the unit and translate to Watts if needed

suneworld, 2022-01-28:
More info:
This is a screenshot from my Grafana graph showing the difference between reported W and kW. 
It looks like it changes from XXX to X.XX when exceeding 1000 W. 

[image]

hultenvp, 2022-01-28:
Got a fix for test in
https://github.com/hultenvp/solis-sensor/tree/56_Fix_variable_power_unit

Master has changed considerably since last release, so ensure you copy all files from that branch.
I have no reference output from a kW scenario, but if the unit is reported as 'kW' then it should work.

hultenvp, 2022-01-30:
@suneworld, can you confirm the fix works?

suneworld, 2022-01-30:
Hi HultenVP! 
I updated to the latest release (v2.0.1). I noticed the changelog included the fix so I assume this can be considered a proper test of the fix?

The fix did not work I'm afraid. The reported value did not change from W to kW. My output was over 1kW today so I had a chance to test.
I wrote my own (dirty) workaround however. The logic is as follows:
If value < 10 then remove punctuation and multiply with 1000
If value >10 do nothing
If value = 20.0 change value to 0

I wrote this in Node-Red and I confirmed today that this logic works.

hultenvp, 2022-01-30:
Hi @suneworld, no, you really need to install the code from the branch mentioned 3 comments up.
The only file changed is ginlong_api.py, so copy that file from the branch to your install and it should do fine.

Thanks!

suneworld, 2022-01-30:
Ah, sorry!
I have now downloaded and copied the file to my install. Will let you know tomorrow how it goes if the sun is up :) :)

hultenvp, 2022-01-31:
Hey @suneworld, does the fix work? Preparing a new patch release today, hope to take this along.

suneworld, 2022-01-31:
Hi!
My output has not exceeded 1kW today so the fix has not been tested I'm
afraid!
Will let you know.

-
Tryllekunstner Sune Alexandersen Narud
Øvre Måsan, Asker
Tlf 924 88 111
facebook.com/sunenarud
sunenarud.no <http://www.sunenarud.no>

bckmrk, 2022-01-31:
I have also tried it but not did exceeded 1 kW today...

hultenvp, 2022-01-31:
No worries, hopefully the weather improves enough tomorrow. 

Cheers

hultenvp, 2022-02-02:
Any updates on the fix?

bckmrk, 2022-02-02:
Too much snow on my panels, still! 🤪

suneworld, 2022-02-02:
Same here!
Only 0.1kWh total production today and faaaaar away from 1000w at any time
:)

hultenvp, 2022-02-02:
Snow? What's that? ;-) Only 6 months of autumn here. Year total 16kWh so far....

Hopefully one of the other people reporting can help confirming the fix in the test branch works:
https://github.com/hultenvp/solis-sensor/tree/56_Fix_variable_power_unit

FYI: I just updated the branch by merging in all stability improvements from master.

bckmrk, 2022-02-02:
> Snow? What's that? ;-) Only 6 months of autumn here. Year total 16kWh so far....
>
> Hopefully one of the other people reporting can help confirming the fix in the test branch works: [...]

:-) Nice. 26 kWh year total here so far, completed installation 22 jan. Your 16 kWh sounds... few, compared to my 26 + snow?

hultenvp, 2022-02-03:
Only have one string of 1620Wp on that inverter. It's oriented east and has some apartments blocking the sun for a few hours in winter season. A particularly sunless month doesn't help either. One more month of patience ;-)

We need someone from Australia complaining about this issue.

hultenvp, 2022-02-04:
I took the risk and released the fix in v2.0.6, let me know if you still experience issues

closed 2022-02-04 by hultenvp

bckmrk, 2022-02-05:
Can confirm it’s working as expected now. 👍🏼
[image]

suneworld, 2022-02-07:
Hi!
I can also confirm that it is working now. Well done!

-
Tryllekunstner Sune Alexandersen Narud
Øvre Måsan, Asker
Tlf 924 88 111
facebook.com/sunenarud
sunenarud.no <http://www.sunenarud.no>