hultenvp/solis-sensor #64: FR: Inverter as a device ?

LucidityCrash, 2022-01-25:
Would it be possible to make it so that each inverter shows up as a device with a number of sensor entities associated with it rather than just the sensor entities by themselves ?

hultenvp, 2022-01-26:
https://developers.home-assistant.io/docs/device_registry_index/

Looks like a good one. I'll have a look, but will have to wait a bit I'm afraid. Bit busy.

hultenvp, 2022-01-26:
Hey @LucidityCrash 

I couldn't resist taking a look at the device registry thing.
Looks like a pretty straightforward thing to implement.

We could keep it simple and inly introduce device "inverter" or we could go with a bit more elaborate device structure and use the via_device:

```
inverter -+--- battery
          |--- string1
          |--- string2
          |--- ...
          |--- stringx

```

It would give some structure to all sensors. Battery definitely makes sense in my view, string devices could be a nice way to cluster all DC sensors. Phase is a bit difficult to see as a separate device, would just have that under inverter, same with grid.

What do you think?

LucidityCrash, 2022-01-26:
hmmm interesting ... I was originally just thinking of just a single device (inverter) that all the sensors would be listed under so you can see related sensors etc.  The idea of breaking those into groups is interesting but does it really gain you anything  over just having everything under inverter ?

hultenvp, 2022-01-26:
For the battery I see the value. It's an optional and separate part only present in hybrid systems. For the strings I admit it's debatable. Battery device and related sensors would look a bit like this screenshot, but then "connected via inverter X"

[image]

Apart from the device overview under configuration I see no usage of the device registry. Perhaps there are custom cards, but for now it looks like a feature that is not used to it's potential.

hultenvp, 2022-02-06:
Look s like I first need to implement the config flow:
https://community.home-assistant.io/t/how-to-add-devices-to-device-registry/160537/3

Dduck112, 2022-04-28:
Hello sir,
I just started with HA and own a Growatt and a solis inverter.
For your info ill show you the growatt setup:
Its an intergration with a fiew devises and sensors. My setup is in dutch.

the integration taile: 
[image]
the devices: 
[image]
the sensors for dev1:
[image]
the sensors for dev 2:
[image]

hultenvp, 2022-10-21:
Done.
Device registry only works for integrations with a config entry (=config flow).
Config flow and device now added.

closed 2022-10-21 by hultenvp