hultenvp/solis-sensor #65: Temperature Sensor stopped working

dynarida, 2022-01-26:
"sensor.my_solis_inverter_temperature" has stopped reporting temperature as of a few days ago. The "state" is shown as unknown, other measurement sensors are ok. I use temperature for automation of supplementary cooling fan and has worked flawlessly for the previous 5 weeks. The Solis integration is up to date (v1.0.4)

hultenvp, 2022-01-26:
Hi,

That's odd.
Can you please add your configuration, logs and ha version?

Cheers

conallob, 2022-01-26:
I'm seeing the same problem with multiple metrics, including temperature and the daily power.

The developer console is telling me "Last Updated: 26 January 2022, 09:41:46" which is when I upgraded this plugin to v1.0.4 and upgraded to Home Assistant 7.2

I will grab logs shortly

conallob, 2022-01-26:
Actually, data is appearing now after forcing multiple restarts in  row.

hultenvp, 2022-01-26:
Hi @conallob 

I've seen with another user that the http requests I'm doing for the auto detect can timeout when your system is heavy on CPU during startup. Added some retries to mitigate, but maybe in your case it's still too low. You might want to increase the max retries in server.Py and see if that helps.

dynarida, 2022-01-26:
Didn't notice before, but can also confirm multiple Solis sensors are being affected. Just had a look at Grafana, and data acquisition seems erratic or non existent. Problems started around 24/1/2021 and that was about the time I upgraded to the previous version 1.0.3? prior to then upgrading v1.0.4. Was on Home Assistant 7.1 then, but now 7.2.
Where do I find the config & logs you want??

hultenvp, 2022-01-27:
The config is the one you entered in configuration.yaml to get the solis integration running, similar to the one mentioned in the readme. I'm only interested in the solis config part, not the rest of your configuration.yaml.
For the logs, I means the HomeAssistant logs.

If you add the following lines to your config and restart then the debug output of the solis integration shows up in HA logs. Please either copy the log in here or add the log file that can be found in the same directory as your configuration.yaml. Don't forget to remove anything you consider privacy sensitive

```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```

LucidityCrash, 2022-01-27:
Strange ... I've been using the soliscloud branch and have been getting data from both soliscloud.com and m.ginlong.com and both have been fine for me (when I've not been mucking about with the soliscloud code 😁)

hultenvp, 2022-01-27:
I really think this is the auto discovery process that takes place during startup.
With some the startup phase is CPU heavy, either because of the system they run on or because of the integrations they use.
In those cases the HTTP requests time out and retries are needed. Maybe in the case of his issue the MAX retry of 3 is just too low. With Conall's setup it looks he's on the edge where sometimes 3 retries just suffices but most of the time not.

henrikeri, 2022-01-27:
@hultenvp 

I seem to be having the same issue now that I upgraded to the latest version. I have three inverters and they are all displaying 0degC.
From the looks the only mention of temperature (running debug mode) are what I have attached below.

`
2022-01-27 11:23:31 DEBUG (MainThread) [custom_components.solis.sensor] Creating inverterstate (160772216070011)
2022-01-27 11:23:31 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute state for inverter 160772216070011
2022-01-27 11:23:31 DEBUG (MainThread) [custom_components.solis.sensor] Creating timestamponline (160772216070011)
2022-01-27 11:23:31 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute timestampOnline for inverter 160772216070011
2022-01-27 11:23:31 DEBUG (MainThread) [custom_components.solis.sensor] Creating timestampmeasurement (160772216070011)
2022-01-27 11:23:31 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute timestampUpdate for inverter 160772216070011
2022-01-27 11:23:31 DEBUG (MainThread) [custom_components.solis.sensor] Creating temperature (160772216070011)
2022-01-27 11:23:31 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute inverterTemperature for inverter 160772216070011
`

`
2022-01-27 11:23:31 DEBUG (MainThread) [custom_components.solis.sensor] Creating inverterstate (010CF1216300005)
2022-01-27 11:23:31 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute state for inverter 010CF1216300005
2022-01-27 11:23:31 DEBUG (MainThread) [custom_components.solis.sensor] Creating timestamponline (010CF1216300005)
2022-01-27 11:23:31 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute timestampOnline for inverter 010CF1216300005
2022-01-27 11:23:31 DEBUG (MainThread) [custom_components.solis.sensor] Creating timestampmeasurement (010CF1216300005)
2022-01-27 11:23:31 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute timestampUpdate for inverter 010CF1216300005
2022-01-27 11:23:31 DEBUG (MainThread) [custom_components.solis.sensor] Creating temperature (010CF1216300005)
2022-01-27 11:23:31 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute inverterTemperature for inverter 010CF1216300005
`

`
2022-01-27 11:23:31 DEBUG (MainThread) [custom_components.solis.sensor] Creating inverterstate (010CF1216300029)
2022-01-27 11:23:31 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute state for inverter 010CF1216300029
2022-01-27 11:23:31 DEBUG (MainThread) [custom_components.solis.sensor] Creating timestamponline (010CF1216300029)
2022-01-27 11:23:31 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute timestampOnline for inverter 010CF1216300029
2022-01-27 11:23:31 DEBUG (MainThread) [custom_components.solis.sensor] Creating timestampmeasurement (010CF1216300029)
2022-01-27 11:23:31 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute timestampUpdate for inverter 010CF1216300029
2022-01-27 11:23:31 DEBUG (MainThread) [custom_components.solis.sensor] Creating temperature (010CF1216300029)
2022-01-27 11:23:31 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute inverterTemperature for inverter 010CF1216300029
`

hultenvp, 2022-01-27:
@henrikeri : Can you confirm _only_ inverterTemperature is reported as 0C? Are all other sensors fine?

henrikeri, 2022-01-27:
@hultenvp: Correct. There are a few of the PV sensors reporting static values, but I assume that's because not all my inverters are 3-phase or multi string(?)

On thing that might be worth looking into, should time be reported in human readable format instead of epoch (i can start a new request if you'd like me to?)

martin-92, 2022-01-27:
For me the same issue. The temp sensor is nog working but al the other sensors are working fine.
Maybe good to know, i am missing the temperature graph also in the Solis app...

dynarida, 2022-01-27:
Exported my data from Solis website and confirm that Temperature stopped being recorded 25th Jan 16:18 EST Australia

LucidityCrash, 2022-01-27:
> Exported my data from Solis website and confirm that Temperature stopped being recorded 25th Jan 16:18 EST Australia

As in it is not showing on the exported data either ?

hultenvp, 2022-01-27:
I'll make a test version tomorrow with extra debug on the temperature. I might be missing something obvious, but from the code I see no particular reason why it should fail only on temp.

In the meantime I have a few sanity check questions to all reporting this issue:
- Sanity check on locale: To all who are reporting this issue: Are you using Celsius or Fahrenheit on the ginlong portal? 
- Can you double check in your HA history there's not another sensor that actually reports the temperature? Maybe for some reason I overlooked the uniqueId of the temperature changed and HA created a new temp sensor.

conallob, 2022-01-27:
I am seeing stale data for multiple sensors, including sensor.solis_inverter_energy_today

Restarting Home Assistant will force sensors to be updated, but it appears the timeout issues are ongoing, not just at startup

martin-92, 2022-01-27:
i use Celcius.

But i think the problem is on the Solis side. They removed the temperature sensor from their app and logging website.

hultenvp, 2022-01-27:
[image]

Not with me.
The plot deepens.

LucidityCrash, 2022-01-27:
Or me

martin-92, 2022-01-27:
> 

But do you also see the temperature on the website? https://m.ginlong.com/

LucidityCrash, 2022-01-27:
So what region is every one in - perhaps this is region based ?  I'm in the UK and still getting Temps.

martin-92, 2022-01-27:
That is really strange. i have access to multiple plants and locations in one country. All of them are missing the temperature.

hultenvp, 2022-01-27:
I do not see temp on the ginlong site. Never looked at it, only accessed via HA so unsure if it ever was there. Looking at desktop version of site, not using the app.

LucidityCrash, 2022-01-27:
I don't see it anywhere distplayed on the web pages just in the App, I have to export the data from the inverter details page or get the data from the API.  For me it is column AR in the downloaded spreadsheet.

dynarida, 2022-01-27:
From the Ginlong website, my exported detailed data for days up to and including the 25th Jan, shows a column for Inverter Temperature, but the reported temperature value suddenly stops midway through the day on the 25th Jan. For subsequent days the Inverter Temperature column is now missing from the data export. So the problem may well be on the Ginlong side?  Location: Australia

hultenvp, 2022-01-28:
If the problem is server side then everting code and config to v0.4.1 would still see the issue.
Can someone confirm?
Also interested in @conallob . I think that's a separate issue

hultenvp, 2022-01-29:
I'm releasing a new version today. That version will not solve this issue.

To be able to resolve this issue I need either confirmation that the issue is on the backend and v0.4.1 also no longer delivers temperature data, or I need logs from the version in this branch to be able to determine root cause:
https://github.com/hultenvp/solis-sensor/tree/65_Temperature_sensor_stopped_working

dynarida, 2022-01-29:
Temperature sensor just started reporting again during 29th Jan all by itself without any intervention from me. That was before I had upgraded to v2.01. Still working on v2.01. So it seems there was a problem on the server side??

hultenvp, 2022-01-30:
I'm going to close this issue as not a bug. Feel free to reopen if there is reason to believe there's an issue with the integration.

Thanks for all the input

closed 2022-01-30 by hultenvp

dynarida, 2022-01-31:
Thx for your help