hultenvp/solis-sensor #66: Today and Total production not updating

sheminasalam, 2022-01-26:
After the recent updates today and total are not updating in my Sofar solar inverter. These are marked as unknown.

hultenvp, 2022-01-26:
Hi, I'll need some more info to find the cause.

Please post your configuration and log output as well as versions of the integration and home assistant

Thanks

sheminasalam, 2022-01-26:
Hi, after trying to sort it out and failing I converted my setup to your own direct wifi logger method and uninstalled this integration. But I  am sharing the integration that I have been using.

```
  - platform: solis
    name: "Sofar Inverter"
    portal_domain: "home.solarman.cn"
    portal_username: "rth@gmail.com"
    portal_password: !secret sofar
    portal_plant_id: "1456096"
    inverter_serial: "SH1ES0****G092"
    sensors:
      actualpower:
      energytoday:
      status:
      temperature:
      dcinputvoltagepv1:
      dcinputcurrentpv1:
      acoutputvoltage1:
      acoutputcurrent1:
      energylastmonth:
      energythismonth:
      energythisyear:
      energytotal:
```

I tried the latest 1.0.4 integration with the latest HA. Even with the 1.0.0 integration, the sensor was only getting updated not regularly which was seen as gaps in the solar energy production graphs in the energy tab. Version 0.0.4 was working great for me.

hultenvp, 2022-01-26:
2 scenarios you can explore:
Remove all sensors and inverter serial from the config. The integration now auto detects the sensors and if all goes well then they should show up in the history. If you change log level to debug then it will show in the log what it detects.

Since you were in such an old version before: know that I've introduced unique id's to stay compatible with home assistant. As a consequence any old sensors are replaced with new ones. It would show up as for example Solis. Energy_today2. The old ones will then be unknown as they are no longer updated. 

If nothing of the above works then please provide debug logging. Perhaps there are little differences between solarman and Solis that I need to take care of

sheminasalam, 2022-01-26:
I tried your first option and now it seems to update. Will let you know if there is any gap in updates lie before.

closed 2022-01-26 by sheminasalam