hultenvp/solis-sensor #68: *inverter_ac_output_total_power reports kW as W

suneworld, 2022-01-27:
Hi! 
After the last update the AC Output Total Power reports the Watts as Kilo Watts.

[image]
[image]

My workaround locally is to create a sensor that multiplies the value with 1000 - but maybe this should be fixed in the next release?

`- platform: template
  sensors:
    power_production_now:
      icon_template: "hass:flash"
      unit_of_measurement: "kW"
      value_template: >-
        {{ (states('sensor.my_solis_inverter_ac_output_total_power')|float(0)*1000)|round(0)}}
`

LucidityCrash, 2022-01-27:
See issue #56 😁

suneworld, 2022-01-27:
Yes, duplicate. Closing.

closed 2022-01-27 by suneworld