hultenvp/solis-sensor #69: Expose Startup Retry Limit as a Configuration Option

conallob, 2022-01-27:
A follow up to the conversation in  #65, it would be nice to override the retry default via a config parameter, instead of making a code change.

I will send a pull request to do this (once I work out how)

hultenvp, 2022-01-28:
Had some spare time.

Got a fix here. 
https://github.com/hultenvp/solis-sensor/blob/69_expose_retry_parameters-_as_config/custom_components

martin-92, 2022-01-30:
Still got this issue with the last version:

Logger: homeassistant
Source: custom_components/solis/service.py:151
Integration: solis (documentation, issues)
First occurred: 11:26:33 (2 occurrences)
Last logged: 11:26:33

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solis/service.py", line 171, in async_update
    await self.update_devices(data)
  File "/config/custom_components/solis/service.py", line 151, in update_devices
    self._subscriptions[serial][INVERTER_STATE].measured
KeyError: 'state'

hultenvp, 2022-01-30:
Hi @martin-92, please open a bug for this. This issue is an improvement and has nothing to do with the crash you observe.  Don't forget to add debug output. The crash shouldn't happen, but without more data I'm unable to find the root cause.

Thanks

hultenvp, 2022-01-30:
@conallob: I can add the retries and delay to the configuration for now, but it conflicts with the config flow work I'm doing. All intrgrations are supposed to move to the config flow, however config flow does not support changing the config. That would mean you can set delay and max retries once, but never change it.

Alternatively I isolate the discovery retry config in one file, then it's not in configuration.yaml, but still separate. Would that work for you?

hultenvp, 2022-01-31:
I've changed the discovery.
It now retries indefinite until successful with a delay of 60secs between every delay without hanging up setup.
Implemented in v2.0.3

closed 2022-01-31 by hultenvp

conallob, 2022-02-01:
Apologies, I couldn't find any time to come back to this for a few days.

Glad it got implemented, I'll revert my pull request and keep an eye out for other simple things I can take on

hultenvp, 2022-02-01:
No prob, hope this solution works for you