hultenvp/solis-sensor #75: Problem with version 2.0.0

cpuks, 2022-01-29:
On 1.0.4 everything works fine but 2.0.0 gives me such error:
`Logger: homeassistant.components.sensor
Source: custom_components/solis/ginlong_base.py:61
Integration: Sensor (documentation, issues)
First occurred: 12:08:35 (2 occurrences)
Last logged: 12:08:35

Error while setting up solis platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solis/sensor.py", line 175, in async_setup_platform
    sensors = await async_autodetect(service)
  File "/config/custom_components/solis/sensor.py", line 90, in async_autodetect
    capabilities = await inverter_service.discover()
  File "/config/custom_components/solis/service.py", line 99, in discover
    capabilities = await self._do_discover()
  File "/config/custom_components/solis/service.py", line 120, in _do_discover
    capabilities[inverter_serial] = data.keys()
  File "/config/custom_components/solis/ginlong_base.py", line 61, in keys
    available_measurements.remove(INVERTER_STATE)
NameError: name 'INVERTER_STATE' is not defined`

henrikeri, 2022-01-29:
I see the exact same error message. Three inverters on my account if that matters.

cpuks, 2022-01-29:
I've got two but it does not matter - even with one config entry I face this error, anyway good that I'm not alone - reverted to 1.0.4 and everything works fine.

hultenvp, 2022-01-29:
Merge conflict. One line disappeared.

hultenvp, 2022-01-29:
Fixed, refresh integration details in HACS and upgrade to v2.0.1

closed 2022-01-29 by hultenvp