hultenvp/solis-sensor #76: 2.0.1 - Energy from Yesterday reported

henrikeri, 2022-01-30:
**Describe the bug**
2.0.1 Still reports energy from yesterday. However, today only energy from 1 repeater was reported if this is of any help.

**Config**
sensor:
  - platform: solis
    name: "Inverter"
    portal_domain: "m.ginlong.com"
    portal_username: !secret solisUser
    portal_password: !secret solisPass
    portal_plant_id: !secret solisPlant

**Versions**
HA Version: core-2022.2.0b3, Home Assistant OS 7.2
HACS 1.20.0
Solis 2.0.1

**Debug logs**
No log issue reflects what is happening.
 
**Screenshots**
[image]

hultenvp, 2022-01-30:
Hmmm, this is....interesting. The fix works on my inverter, but apparently still some corner case I overlooked. More leads are welcome so that I can try to reproduce.

Siwek666, 2022-01-30:
Works perfect for me i think with solarman platform integration.. Thanks so much.

Versions
HA Version: core-2021.12.10, Home Assistant OS 7.2
HACS 1.20.0
Solis 2.0.1

[image]

hultenvp, 2022-01-30:
Ugghhh, I think I found it. Line 153 in service.py contains a wrong test; '<' should have been '>'. As a result the test evaluates wrong and returns for one measurement still the energy of yesterday. No clue why the problem went away with me and you still see it.
Running tests now.

henrikeri, 2022-01-30:
I have changed the line as well. Will report results tomorrow around noon when it is up and running. It also only seems to get 1 of the DC lines PV1, not PV2 of my multi string inverter. Is this related to the same?

hultenvp, 2022-01-30:
Thanks looking forward to the result. I tested it, looks to be okay now, but I want to see if it doesn't break anything unexpected tomorrow when the sun comes up 

Your other issue seems unrelated. Could you let a debug log and create a new issue for that?

Cheers

henrikeri, 2022-01-31:
@hultenvp, the change seems to work at my end as well. It showed up as intended after sunrise this morning. As for the issue with PV2, it is once more showing.....

hultenvp, 2022-01-31:
Thanks for letting me know this fix works, please open an issue for the PV2, but please add debug logging as I'm only having one string here.

hultenvp, 2022-01-31:
Released in v2.0.2. Let's pray it is now really fixed

closed 2022-01-31 by hultenvp

cpuks, 2022-02-06:
Just had this issue again on 2.0.6 - whole yesterday production added to todays morning.

[image]

kazikskrz, 2022-03-28:
I'm having the same here. Every few days I get production value from day before.
version 2.1.0
HA 2022.3.5
[image]
[image]
[image]