hultenvp/solis-sensor #77: Solis Sensors Stop Updating at Midnight Each Day

conallob, 2022-01-31:
**Describe the bug**
Solis sensors for my system stop updating every night at midnight. The next day, sensors such as sensor.solis_inverter_energy_today were last updated around midnight the night before. Eg 30 January 2022, 23:58:06

To mitigate, I have to restart my Home Assistant device (restarting Core, Supervisor or a config restart are not enough)

**Config**
Your anonymized config entry

**Versions**
HA Version an deployment core-2021.12.10 on a RPi 4 4GB
HACS version: 1.21.0
Integration version: v2.0.1

**Debug logs**
Add to configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**
If applicable, add screenshots to help explain your problem.

hultenvp, 2022-01-31:
Hi @conallob,

Your Solis switches off for the night and starts up in the morning. This is normal behaviour.
In the morning it should pick up again. 
Energy today is a special case, as it does not reset at midnight and messes up the energy dashboard, so there's a reset in place to reset to 0 at midnight.
If it does not pick up again in the morning then something's wrong. Please attach debug info if that's the case.

conallob, 2022-02-01:
This repeated again overnight.

My logs state:

First occurred: 31 January 2022, 23:27:58 (16 occurrences) 
Last logged: 31 January 2022, 23:58:11

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solis/service.py", line 174, in async_update
    await self.schedule_update(update)
TypeError: object NoneType can't be used in 'await' expression

2022-01-31 23:48:06 INFO (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: [Recacted]
2022-01-31 23:48:07 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 2 minutes.
2022-01-31 23:48:07 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/config/custom_components/solis/service.py", line 174, in async_update
await self.schedule_update(update)
TypeError: object NoneType can't be used in 'await' expression
2022-01-31 23:50:07 INFO (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: [Recacted]
2022-01-31 23:50:08 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 2 minutes.
2022-01-31 23:50:08 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/config/custom_components/solis/service.py", line 174, in async_update
await self.schedule_update(update)
TypeError: object NoneType can't be used in 'await' expression
2022-01-31 23:52:08 INFO (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: [Recacted]
2022-01-31 23:52:09 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 2 minutes.
2022-01-31 23:52:09 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/config/custom_components/solis/service.py", line 174, in async_update
await self.schedule_update(update)
TypeError: object NoneType can't be used in 'await' expression
2022-01-31 23:52:12 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 134, in _handle_refresh_interval
await self._async_refresh(log_failures=True, scheduled=True)
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 265, in _async_refresh
update_callback()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 505, in async_write_ha_state
self._async_write_ha_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 541, in _async_write_ha_state
extra_state_attributes = self.extra_state_attributes

Hopefully that is of some help

conallob, 2022-02-02:
Thank you for the quick fix and release!

I've upgraded to v2.0.4, but I'm still seeing the same issue, but on a different line number in service.py

First occurred: 00:00:02 (1 occurrences) 
Last logged: 00:00:02

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solis/service.py", line 168, in async_update
    await self.update_devices(data)
  File "/config/custom_components/solis/service.py", line 148, in update_devices
    self._subscriptions[serial][INVERTER_STATE].measured
KeyError: 'state'

Full log:

2022-02-02 00:00:02 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/config/custom_components/solis/service.py", line 168, in async_update
await self.update_devices(data)
File "/config/custom_components/solis/service.py", line 148, in update_devices
self._subscriptions[serial][INVERTER_STATE].measured
KeyError: 'state'

hultenvp, 2022-02-02:
I'm really struggling to see how this can still happen. There's an explicit check in the code that ensure the service only gets data when essential keys, like 'state' are in de data. I'll do a proper code review, but might take a while. In the meantime, could you try to force a reinstall? HACS->Solis->3 vertical dots->Redownload. Probably won't change anything, just to be sure.

conallob, 2022-02-02:
I've re-downloaded it now and have verified pycache was deleted too.

Is INVERTER_STATE in any way related to the Inverter status sensor? The Developer Tool console reveals sensor.solis_inverter_status is currently the only solis sensor without a value (unknown)

hultenvp, 2022-02-02:
Thanks,
Let me know if it helped

Status is deprecated. INVERTER_STATE maps to Solis.state. value of 2 means inverter off, 1 inverter on. Must always be there.

hultenvp, 2022-02-02:
I reviewed the code and fixed some issues. Release them in 2.0.5. Don't think it solves the root cause.

Your instance crashes because the state sensor is not/no longer subscribed. Cannot see from the code review why that is.
Discovery would fail and retry when state is not found in the data, but the crash you see is during normal operation, so the state was found, sensor got installed and then somehow it gets removed?

Let's see what happens with the new release, if it gives some leads towards the root cause.

conallob, 2022-02-03:
Data is starting to appear today without any intervention from me.

I'm happy to confirm that v2.0.5 does appear to have fixed this issue as well.

hultenvp, 2022-02-03:
Thanks for letting me know Conall.

I'll leave this issue open for another 24 hrs just in case.

conallob, 2022-02-04:
Confirmed, second day in a row without restarting (although I did upgrade to HA Core 2022.2.0 within the last 24 hours).

I'm happy to declare this issue fixed

conallob, 2022-02-04:
Thank you very much!

closed 2022-02-04 by hultenvp