hultenvp/solis-sensor #78: Error while setting up solis platform for sensor

jahpster, 2022-01-31:
after the last update the solis integration stopped working.

2022-01-31 12:14:43 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up solis platform for sensor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
await asyncio.shield(task)
File "/config/custom_components/solis/sensor.py", line 175, in async_setup_platform
sensors = await async_autodetect(service)
File "/config/custom_components/solis/sensor.py", line 90, in async_autodetect
capabilities = await inverter_service.discover()
File "/config/custom_components/solis/service.py", line 99, in discover
capabilities = await self._do_discover()
File "/config/custom_components/solis/service.py", line 118, in _do_discover
data = await self._api.fetch_inverter_data(inverter_serial)
File "/config/custom_components/solis/ginlong_api.py", line 253, in fetch_inverter_data
self._post_process()
File "/config/custom_components/solis/ginlong_api.py", line 300, in _post_process
float(self._data[INVERTER_TIMESTAMP_ONLINE])/1000
KeyError: 'timestampOnline'

hultenvp, 2022-01-31:
Can you give me some more info?
- What was your previous version?
- Does this happen all the time or is it sometimes successful after a reset?
- Can you run with debug on and attach that log?

jahpster, 2022-01-31:
previous version was 2.0
this happens all the time after updating to the latest version.
I had some problems with 2.0 but after a few restarts everything worked again

2022-01-31 14:41:32 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solis
2022-01-31 14:41:44 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error:
2022-01-31 14:41:44 INFO (MainThread) [custom_components.solis.service] Discovery failed, retry attempt #1
2022-01-31 14:41:57 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-01-31 14:41:57 INFO (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 110E30181100****
2022-01-31 14:41:58 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up solis platform for sensor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
await asyncio.shield(task)
File "/config/custom_components/solis/sensor.py", line 175, in async_setup_platform
sensors = await async_autodetect(service)
File "/config/custom_components/solis/sensor.py", line 90, in async_autodetect
capabilities = await inverter_service.discover()
File "/config/custom_components/solis/service.py", line 99, in discover
capabilities = await self._do_discover()
File "/config/custom_components/solis/service.py", line 118, in _do_discover
data = await self._api.fetch_inverter_data(inverter_serial)
File "/config/custom_components/solis/ginlong_api.py", line 253, in fetch_inverter_data
self._post_process()
File "/config/custom_components/solis/ginlong_api.py", line 300, in _post_process
float(self._data[INVERTER_TIMESTAMP_ONLINE])/1000
KeyError: 'timestampOnline'
2022-01-31 14:42:29 DEBUG (MainThread) [custom_components.hacs] <Integration hultenvp/solis-sensor> Getting repository information
2022-01-31 14:42:31 DEBUG (MainThread) [custom_components.hacs] <Integration hultenvp/solis-sensor> Running checks against v2.0.1

hultenvp, 2022-01-31:
My assumption is that the server responds with a payload that does not contain the expected data.
I'll check tonight if I at least can make it robust against false data and let's see if it then gets successful during retry.

hultenvp, 2022-01-31:
Hi @jahpster: Give v2.0.2 a spin. It should at least no longer crash.
Let me know if it helps.

jahpster, 2022-01-31:
thanks for you fast response and all your work!
I did an update and it doesn't crash anymore but i don't get any data or sensors.
I get the same errors over and over again.

2022-01-31 19:46:34 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-01-31 19:46:35 INFO (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 110E30181100
2022-01-31 19:46:35 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes.
2022-01-31 19:47:24 INFO (MainThread) [charset_normalizer] ascii passed initial chaos probing. Mean measured chaos is 0.000000 %
2022-01-31 19:47:24 INFO (MainThread) [charset_normalizer] ascii is most likely the one. Stopping the process.
2022-01-31 19:47:35 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-01-31 19:47:36 INFO (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 110E30181100
2022-01-31 19:47:36 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes.
2022-01-31 19:48:37 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-01-31 19:48:38 INFO (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 110E30181100
2022-01-31 19:48:38 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes.
2022-01-31 19:49:39 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-01-31 19:49:40 INFO (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 110E30181100
2022-01-31 19:49:40 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes.

hultenvp, 2022-01-31:
That's what I was afraid of. Something is different in the responses you get from the server.
Can you uncomment on line 259 in ginlong_api.py: _LOGGER.debug("Payload = %s", payload).
It will then spit out the server response in the debug log. This helps me analyzing what's different with your setup.

Thanks

jahpster, 2022-01-31:
[solis.log](https://github.com/hultenvp/solis-sensor/files/7973622/solis.log)

jahpster, 2022-01-31:
[solis_e.log](https://github.com/hultenvp/solis-sensor/files/7973639/solis_e.log)

hultenvp, 2022-01-31:
You miss the receiveTimestamps in your response.
I'll make a fix.

hultenvp, 2022-01-31:
Fixed in v2.0.3. Feel free to reopen when you still have issues

closed 2022-01-31 by hultenvp

jahpster, 2022-01-31:
Fantastic!
Thank you!