hultenvp/solis-sensor #84: Not getting data from the server after a period of time

Mattis0223, 2022-02-01:
**Describe the bug**
I´m not getting updates from the server after a short period of time. At least not for the current output. The integration works for about 30 min after installing. I'm new to HA and GitHub ... hoping I'm not asking questions that has been answered before.

[image]
[image]

**Config**
  - platform: solis
    name: "My Solis Inverter"
    portal_domain: "m.ginlong.com"
    portal_username: "*********@gmail.com"
    portal_password: "***********"
    portal_plant_id: "******"      
 

**Versions**
HA Version an deployment core-2021.12.10 on a RPi 4 4GB
HACS version: 1.21.0
Integration version: v2.0.1

**Debug logs**
Add to configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

hultenvp, 2022-02-01:
Hi @Mattis0223 . Thanks for using my integration. 
Please upgrade to 2.0.4. v2.0.1 still had some stability issues.
Note that sensors do not update if there's nothing to update.

Mattis0223, 2022-02-01:
sry ... I am at 2.04 and thank you for this integration.

An app that I'm using is getting data

[image]
.
Meanwhile 

[image]

hultenvp, 2022-02-01:
I see what you mean, thanks for the clarification. This is a known issue, see issue #56. When power produced goes up the server switches from reporting in Watts to reporting in kiloWatts. The integration does not check for that and starts reporting lower values as you see in your screenshot 3.46W instead of 3.46kW.
There is a fix under test and you can find a link to the proposed fix in the issue #56, but since you're new to HA best might be to wait it out. Hopefully today or tomorrow we have the fix released.

For now I close this one as a duplicate to keep the discussion centralized. Feel free to contribute in issue #56

closed 2022-02-01 by hultenvp