hultenvp/solis-sensor #85: Incorrect Power conversion

henrikeri, 2022-02-02:
**Describe the bug**
Power generation shows 2.6 W whilst it's in fact 2.6 kW. Seems to happen when generation > 1kW

**Config**
Default config, no changes. Using m.ginlong.com

**Versions**
HA Version an deployment: core-2022.2.0b6 on HA OS 7.2
HACS version: 1.21.0
Integration version: 2.0.4

**Screenshots**
Upper inverter:
[image]
Lower right gauge:
[image]

hultenvp, 2022-02-02:
Hey @henrikeri, thanks for reporting. This is a duplicate of issue #56 . There's actually a link to a fix in that issue. It's under test by the submitters in that issue. Happy if you try it out as well.

Cheers

henrikeri, 2022-02-02:
@hultenvp 
Thanks I've added the latest api file. Hopefully we'll have nice enough weather for a test tomorrow.

Cheers :)

hultenvp, 2022-02-03:
Thanks, make sure you copied the changed file into a v2.0.5 release. This to avoid mixing versions.

kazikskrz, 2022-02-03:
sensor.inverter_ac_output_total_power shows 1,02 W instead of 1,02 Kw

hultenvp, 2022-02-04:
Fixed in v2.0.6, let me know if you still experience issues

closed 2022-02-04 by hultenvp

etzka, 2022-02-04:
> Fixed in v2.0.6, let me know if you still experience issues

it worked fine for me in version v2.0.5. Above 1kw showed good in watts

hultenvp, 2022-02-04:
@etzka : Thanks for the screenshot, so working fine it seems!

kazikskrz, 2022-02-07:
Looks good to me now. Many thanks for quick fix.