hultenvp/solis-sensor #91: Wrong data in the morning

mwitlox, 2022-02-07:
**Describe the bug**
Wrong data in the morning. Data from the day before showing up in the morning

**Config**
Solarman portal config m.ginlong.com (Solis inverter, Wifi dongle)

**Versions**
HA Version 2022.x
HACS version last 2 versions. Now 1.22.0
Integration version latest

**Debug logs** < Will do this when needed
Add to configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**
[image]

mwitlox, 2022-02-07:
Tried redownloading, reconfigure, readd in energy dashboard

hultenvp, 2022-02-07:
Hi @mwitlox, not sure from your last comment if the problem is now solved or actually happening on both platforms?
If the problem still exists; could you let me know how many minutes wide the spike is? Hovering over the spike with mouse shows timestamps

Thanks

mwitlox, 2022-02-07:
Hi @hultenvp. Sorry for that. Both portals are working. I still use the m.ginlong.com portal and your program to get the data to HA. Haven't tried the new portal with your program. Maybe better to use the new portal?

Today 8:17:48 - 8:19:50

04/02/2022 same thing but then 8:42:09 - 8:44:10

hultenvp, 2022-02-07:
Not sure if using the other portal will help. The person that helped me with the support of soliscloud is seeing other issues, so switching might mean you trade one issue for another ;-)
I still have one improvement I can make. I'm going to test that it doesn't break anything and if it works I will drop the release later this week.

LucidityCrash, 2022-02-08:
Its worth noting here that my data is available in both systems also

I have a Hybrid inverter with a battery so it never actually goes offline ... The Daily figures sometimes don't reset at midnight but a few seconds/minutes after - on m.ginlong.com that totally breaks the energy use for the day until the next day or until the amount used today exceeds the amount used yesterday.
On soliscloud it fixes itself the next reading but I've yet to switch over to using the soliscloud data to drive my energy dashboard so not sure how it will affect that.

I've found some data inconsistancies in the SolisCloud website and API ... on the website my "Daily energy to grid" and "Daily energy from grid" are always 0 even though the "Total energy to grid" and "Total energy from grid" are increasing.  However the API returns both the correct "Total energy from grid" AND "Daily energy from grid" but NOT the correct "Daily energy to grid" OR "Total energy to grid".  If I look at what the m.ginlong.com shows the "Daily energy to grid" does sometimes show a very little (like 0.1 kWh) and the "Daily energy from grid" is always non zero so the data is there in the backend it just isn't available via SolisCloud for some reason.  I've currently got tickets open with Solis Support but for the Daily energy to grid one they are saying that the export is so low that it is rounding down and hence not showing and to stop wasting their time ... for the daily energy to grid and the reset to 0 at midnight they are just not answering.

hultenvp, 2022-02-08:
Hey @LucidityCrash ,

From our discussion I understood you had issues on the ginlong portal itself, but the way I understand now you actually have issues with the integration messing up your energy dashboard, because energy today rests to 0 after midnight instead of at midnight?

LucidityCrash, 2022-02-08:
no you were correct it is the data stored in m.ginlong.com that is broken ... when I use the SolisCloud API it works much better.

I assume the issue is down to timing the datalogger sends data every 5 minutes (ish) so today for example the first value is reported as being at 00:00:19 but however the reset is worked out the numbers that it has stored in the backend haven't been reset to zero at this point.

for exaple on m.ginlong.com if yesterday I used 15kWh but that isn't reset till after midnight then today it will say 15kWh until I've used 15.1 kWh ... this screws up the energy dashboard but it is only because the data coming in is bad.  With the SolisCloud API when the data resets to 0 after midnight it actually does reset.

m.ginlong.com api :
[image]

SolisCloud api : 
[image]

Looks like the Daily Yield on this one didn't reset till after midnight on SolisCloud either

hultenvp, 2022-02-08:
Indeed, with normal battery-less inverters the daily energy is reset at inverter power-up in the morning. For you, with battery it is reset somewhere just after midnight.
I think I can fix both scenarios. Pretty sure I have @mwitlox case fixed now (cannot reproduce myself, but should be okay). Going to give your scenario a try as well by ignoring any energy today data from the server the first 15 minutes of a day and just return '0'. 
Will run regression test tomorrow, if all passes will release later this week.

LucidityCrash, 2022-02-09:
15 mins seems excessive ... its never going to be more than 5 mins late resetting (this is on SolisCloud ... m.ginlong.com never resets).

mwitlox, 2022-02-10:
A bit more info. Seems to happen every 3 days. Today 8:37:19 - 8:39:11 same thing.

Total value of the day before is given to HA. Then a bit later the total resets en starts counting from 0. But the energy dash stores the value of the day before + counting the new day from the beginning

Maybe this will make things a bit more clear. Screenshots from this morning

[image]
Time 00:01:15
[image]
[image]

Another thing, because the Wifi module is powering on-off-on-off-on... maybe that is also a problem. Here a screenshot from my P1 DSMR Production
[image]

hultenvp, 2022-02-10:
I just release v2.0.7.

Looking at all the data you shared this should fix the issue.
It also should address the issue of @LucidityCrash .

Let me know if it works.

hultenvp, 2022-02-11:
Was it better this morning?

attila123456, 2022-02-11:
I had the same issue, this morning was all good!

mwitlox, 2022-02-11:
[image]
[image]

This morning a normal fresh start :)!

I will watch is for a couple of days because the issue was every 3 days. But i think it's all good now!

@hultenvp good job 🥇 !

bckmrk, 2022-02-11:
I had some weird notch in the diagram this morning. It was cloudy and a slow start resulted in this behavior

[image]

Close up:

[image]

Looks like multiple resets?

hultenvp, 2022-02-11:
@bckmrk : If it were multiple resets then it should be visible in inverter state. You should see toggles between state value 1 and 2 then.

I sometimes feel some ginlong employee is monitoring me and figuring out ways how they can complicate stuff further 🤣

bckmrk, 2022-02-11:
Don't see that. Don't know what it was.. 

[image]

hultenvp, 2022-02-11:
You had some interrupts of service around that time:
[image]

bckmrk, 2022-02-11:
wow, didnt notice.

hultenvp, 2022-02-12:
Closing based on feedback that the fix works

closed 2022-02-12 by hultenvp