hultenvp/solis-sensor #92: Energy Spike reported from Solis Energy Today

bckmrk, 2022-02-08:
Sensor Solis Energy Today reporterd a non existing energy spike that is consistent with yesterdays 'energy total' this morning. I haven't seen it before but it messes up the energy dashboard with a spike there to. Is it just me? 

[image]
[image]

Siwek666, 2022-02-08:
No from yestarday I have this same problem... two days in a row.. [image]

bckmrk, 2022-02-08:
Seems to be a duplicate of #91.

bckmrk, 2022-02-09:
The spike didn't happen today. Yesterday the spike was from 8:25;09 to 8:27;10, so 2 min and 1 sec?   

Great work anyhow :-)