hultenvp/solis-sensor #93: battery and temperature record have different name

darkmordu, 2022-02-08:
For my Sofar inverter, I see that temperature record name is 1to instead of 1df
battery remaining capacity is 3je istead of 1cv

inverter sofar HYD-6000-EP
logger solarman LSW3_15_270A_1.32

using homeassistant 2022.2.3
solis component 2.0.6

thanks for the integration, works great and improves well

hultenvp, 2022-02-08:
Hi @darkmordu 
Looks like you have access to the json payload.
Could you attach an anonymized version to this ticket?

Thanks

darkmordu, 2022-02-11:
here is the json

2022-02-11 17:51:12 DEBUG (MainThread) [custom_components.solis.ginlong_api] Payload = {'result': {'datalogger': {'address': 'FranceAuvergne-Rhône-AlpesIsèreVillard-BonnotCollege Belledonne', 'datalogger': {'commandType': '02', 'gatewayModel': 'LSW-3B3253-C', 'sn': '2304744799', 'type': 3, 'updateDate': 1644589695000}, 'dataloggerSn': '2304744799', 'dataloggerType': '3', 'extend': '{"11cy":"3","_st":"1","_et":"1644601835","a":"02","stat":"0","b":"3128964","c":"54","d":"0","e":"5","f":"60","za":"2304744799","g":"1","h":"LSW3_15_270A_1.32","zb":"A58200104100015F9D5F890284BE2F003600000000000000053C780131014C5357335F31355F323730415F312E3332000000000000000000000000000000000000000000000034EAE7EC3F8C3139322E3136382E302E3130350000000100010A270300FF56312E312E30302E3042000000000000000000000000000000000000000000000000000000000000FF2615","zc":"78.226.238.234:62105","zd":"1644589695","ze":"1644589696","zf":"","l":"49","zg":"00,01","m":"1","83az":"255","n":"1","zh":"47.88.8.200:10000","zi":"01","o":"270A","ruleCode":"010102","p":"1","zk":"1","zl":"0","t":"120","80ak":"V1.1.00.0B","x":"34EAE7EC3F8C","y":"192.168.0.105","80ai":"255","_ut":"1644589720"}', 'firmwareVersion': 'MWXT-LPB100_RELOAD_V1.0.4(2018-05-25)lsw3.hex.hex', 'hardwareVersion': 'LSW-341433T-VH1.0.2(2019-12-19)', 'isDel': 0, 'lat': 45.245527114, 'lon': 5.89731, 'plantId': 1494203, 'plantName': 'maison', 'productModel': 'LSW-3B3253-C', 'state': 1, 'updateDate': 1644589695000}, 'deviceDisplayName': 'Inverter', 'lngLat': '{"add":"FranceAuvergne-Rhône-AlpesIsèreVillard-BonnotCollege Belledonne","lng":5.89731,"name":"Inverter","href":"http://pro.solarman.cn/index/setIndex.do?type=24&param1=1494203","lat":45.245527114,"plantName":"maison","status":1}', 'deviceWapper': {'address': 'FranceAuvergne-Rhône-AlpesIsèreVillard-BonnotCollege Belledonne', 'alarmWapper': {'alarmContext': '124 Battery Low Voltage Protection|124 电池低电压保护', 'alarmId': 360844443, 'code': 'bit123', 'createTime': 1644563980000, 'dataloggerSn': '2304744799', 'description': '{"a":"124 Battery Low Voltage Protection|124 电池低电压保护"}', 'deviceId': 104478090, 'deviceType': 1, 'level': 3, 'sensor': '270A', 'type': 1, 'updateTime': 1644576917000}, 'data': '{"5yd":"0","12aw":"0.00","5ye":"0","12av":"0.00","5yf":"0","12au":"0.00","5yg":"0.00","12at":"0.00","5yh":"0.00","12as":"0.00","5yi":"0.00","12ar":"0.00","5yj":"0","5yk":"0","5yl":"0","1qh":"0.00","5ym":"0","1qi":"0.00","5yn":"0.00","1qj":"0.00","5yo":"0.00","5yp":"0.00","5yq":"0","12az":"0.00","5yr":"0","12ay":"0.00","5ys":"0","12ax":"0.00","5yt":"0","5yu":"0.00","1aa":"0.00","5yv":"0.00","5yw":"0.00","5yx":"0","5yy":"0","5yz":"0","1af":"236.60","1ag":"0.00","1ah":"0.00","1ai":"2.04","1aj":"0.00","1ak":"0.00","5za":"0","1al":"0.00","5zb":"0.00","1am":"0.00","5zc":"0.00","1an":"0.00","5zd":"0.00","1ao":"450.00","5ze":"0","12bx":"0.00","5zf":"0","12bw":"0.00","1aq":"0.00","5zg":"0","12bv":"0.00","1ar":"49.97","5zh":"0","12bu":"0.00","5zi":"0.00","12bt":"0.00","1re":"6","5zj":"0.00","12bs":"0.00","5zk":"0.00","12br":"0.00","5zl":"0","ah":"10602","12bq":"0.00","5zm":"0","12ca":"0.00","5zn":"0","5zo":"0","5zp":"0.00","5zq":"0.00","5zr":"0.00","5zs":"0","12bz":"0.00","5zt":"0","ap":"0000","12by":"0.00","5zu":"0","12bh":"0.00","ar":"0000","5zv":"0","12bg":"0.00","as":"0000","5zw":"0.00","12bf":"0.00","1bc":"207.00","at":"0000","5zx":"0.00","12be":"0.00","1bd":"3.34","1be":"80.15000000000008","au":"0000","5zy":"0.00","12bd":"0.00","1bf":"207.0099999999999","av":"0000","5zz":"0","12bc":"0.00","aw":"0000","12bb":"0.00","ax":"0000","12ba":"0.00","ay":"0000","12bp":"0.00","az":"0000","12bo":"0.00","12bn":"0.00","12bm":"0.00","12bl":"0.00","12bk":"0.00","12bj":"0.00","12bi":"0.00","1bp":"49.97","1bq":"0.00","12cx":"2","1bu":"0.10","1bv":"554.20","1bw":"0.01","1bx":"2.51","12cr":"0507","1by":"0.06999999999999999","1bz":"156.4899999999999","bp":"0000","1ca":"0.09999999999999999","bq":"0000","1cb":"554.27","1a":"263.40","br":"0000","12ci":"0.00","1b":"21.40","bs":"0000","12ch":"0.00","1c":"0.00","bt":"0000","12cg":"0.00","1d":"0.00","bu":"0000","12cf":"0.00","1e":"0.00","bv":"0000","1cf":"0.17","12ce":"0.00","1f":"0.00","bw":"0000","1cg":"0.00","12cd":"0.00","1g":"0.00","1ch":"0.00","12cc":"0.00","1h":"0.00","12cb":"0.00","1i":"0.00","1cj":"440.00","12cq":"0V030000","1j":"0.00","12cp":"0V030011","1k":"0.00","12co":"0.00","1l":"0.00","12cn":"0.00","1m":"0.00","1cn":"736.20","1n":"0.00","1co":"7.03","1cp":"226.80000000000015","1o":"0.00","12ck":"0.00","1cq":"736.0599999999994","1p":"0.00","_":"0C2F","12cj":"0.00","1q":"0.00","1r":"0.00","a":"01","stat":"0","1s":"0.00","b":"270A","1t":"0.00","c":"3137489","1u":"0.00","d":"8579","1v":"0.00","e":"1641460731","1w":"0.00","f":"0001","1cx":"135.50","1x":"0.00","g":"SM2ES160M8L253","1cy":"105.70","1y":"0.00","1cz":"2.09","1z":"0.00","1tm":"0","1to":"25","1tp":"0","1da":"3.12","1db":"54.66000000000002","1dc":"43.15999999999999","ruleCode":"0102010A27","1dd":"135.49","1de":"105.61000000000006","1dk":"0","dt":"1644598220000","1eo":"870.7333333333333","3it":"-420.00","1ez":"10","3jc":"48.70","3jd":"-8.78","1vs":"394.10","3je":"30","3jf":"17","1vw":"22","3ji":"0.00","1vx":"2","1vy":"11","1vz":"17","3jo":"0.00","11dr":"0.00","3jp":"0.00","3jq":"0","1wa":"49","3jr":"0","1wb":"26","1fs":"","3jx":"0.00","3kd":"0.00","3ke":"0.00","3kf":"0","3kg":"0","17as":"0","3km":"0.00","6aa":"0","6ab":"0","6ac":"0","6ad":"0","6ae":"0","6af":"0","3ks":"0.00","6ag":"-4","3kt":"0.00","6ah":"0","3ku":"0","6ai":"0","3kv":"0","6aj [... cut ...] ', '1ou': '0.00', '1ov': '0.00', '1ow': '0.00', '1ox': '0.00', '1oy': '0.00', '5xr': '98', 'a0': '1', '5xs': '0', '5xt': '0', '5xu': '0', '5xv': '0', '5xw': '0', '5xx': '0', '5xy': '0', '5xz': '0.00', '5ya': '0.00', '5yb': '0.00', '5yc': '0'}, 'deviceBasicInfo': [{'key': 'g', 'name': 'SN', 'value': 'SM2ES160M8L253'}], 'deviceVersionInfo': [{'key': '1gv', 'name': 'Hardware version', 'value': 'V001'}, {'key': '2lb', 'name': 'Communication CPU software version', 'value': '0V030012'}]}}, 'state': 1}

hultenvp, 2022-02-12:
Thanks for the JSON dump. This clearly a system version I haven't seen before.
I can add those, but I cannot map them on the original sensors; they will become new/different sensors.
It's a bit busy time, so it will take a few weeks before you have them.

hultenvp, 2022-02-20:
Hi @darkmordu ,

Got a test version for you:
https://github.com/hultenvp/solis-sensor/archive/refs/heads/93_add_solarman_battery_and_temp_sensors.zip

Just replace the files in your solis custom components with the one from the zip.
You should get 2 new sensors for temp and battery capacity.

Let me know if it works.

hultenvp, 2022-02-23:
Hi @darkmordu ,

Does the test version work?

THanks

darkmordu, 2022-02-24:
Hi,
I tried the version today but it does not work (the batt Soc and temperature do no show)
When I look at the code I see no changes in the record values taken, so it is expected.
Thanks

hultenvp, 2022-03-04:
Me bad, I forgot to commit the changes, so indeed nothing to see.

The same branch now contains the changes, can you please retry?

hultenvp, 2022-03-09:
Released in 2.1.0. Let me know if there are still issues

closed 2022-03-09 by hultenvp