hultenvp/solis-sensor #97: Would GRID_TOTAL_ENERGY_USED be better as MWh

LucidityCrash, 2022-02-19:
Seems to me that putting this in kWh  for something that is a lifetime total is probably too small a scale and it might be better as MWh, the same could be said for INVERTER_ENERGY_TOTAL_LIFE, BAT_TOTAL_ENERGY_CHARGED, and BAT_TOTAL_ENERGY_DISCHARGED ?

Thoughts ?

hultenvp, 2022-02-20:
Just checked HA core const file. MWh is a valid unit for sensors. So in theory we could change it. 
Not sure if it affects energy dashboard. I think it is always in kWh and I personally use energy today because it has higher precision.
Also not sure how existing sensors and energy dashboard will respond. Will have to test. If it breaks history then I'd propose not to change it.

LucidityCrash, 2022-02-20:
Surely these would only affect the energy dashboard if you used it as the sensor for one of the things it measures, and as you say using the daily one is much more useful in that situation.  Changing it would probably muck up the history though.  It just seemed to me that some of these life sensors are easily going to be MW, in the space of 2 months I'm at 1.1 MW for GRID_TOTAL_ENERGY_USED,  and probably even GW within the lifespan of the inverter.

hultenvp, 2022-03-11:
I just tried out changing INVERTER_ENERGY_TOTAL_LIFE tot MWh. Sensor nicely changes to new unit and value.
History isn't able to entirely deal with it. It's still kWh and now reports the MWh value as kWh.
[image]
When I zoom in to a time period with only MWh values then the unit is correct.
[image]
That should not be problematic in the end as history is typically only a few days. 
People who also record the data in for example Influx and use Grafana must be careful to change their queries.

Next step is testing energy dash.

hultenvp, 2022-03-12:
Energy dash refuses to consume from a sensor with values with different units.
That means as long as recorder contains values for a sensor in two different units it won't work to use energy total in MWh for energy dash.
It is however unlikely that people uses energy total for energy dash, so how much of a problem is it...