hultenvp/solis-sensor #104: Mixed up sensor values

Xplode85, 2022-03-14:
**Describe the bug**
dc_power_pv3 has the wrong value.
dc_power_pv3 seems to use the value from dc_voltage_pv3 just with the decimal separator moved 1 over.

**Config**
Your anonymized config entry

**Versions**
Integration version: 2.1.0

**Screenshots**
[image]
[image]
[image]

hultenvp, 2022-03-14:
I see your point, however if I review the code then the fields I read are the correct ones.
How is the data on ginlong.com? Is the data wrong there as well?

Xplode85, 2022-03-14:
Thanks for checking.
I checked the website myself and found the problem...but not sure about the cause of the problem.
As you can see here PV3 current is frozen at 0.1A
[image]
I found PV3 current has been stuck at 0.1A since the install of the system.
I guess there is some kind of software problem with the inverter.

closed 2022-03-14 by Xplode85