hultenvp/solis-sensor #105: Error after delete and reinstall

foncubierta, 2022-03-27:
Good morning, first of all comment that the component works great, but because of me I accidentally deleted the hacs component, when reinstalling it it doesn't work, it gives me this error:

Logger: homeassistant
Source: custom_components/solis/ginlong_api.py:353
Integration: solis ([documentation](https://github.com/hultenvp/solis-sensor/), [issues](https://github.com/hultenvp/solis-sensor/issues))
First occurred: 10:59:20 (1 occurrences)
Last logged: 10:59:20

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solis/service.py", line 98, in async_discover
    capabilities = await self._do_discover()
  File "/config/custom_components/solis/service.py", line 115, in _do_discover
    data = await self._api.fetch_inverter_data(inverter_serial)
  File "/config/custom_components/solis/ginlong_api.py", line 269, in fetch_inverter_data
    if self._collect_inverter_data(payload):
  File "/config/custom_components/solis/ginlong_api.py", line 308, in _collect_inverter_data
    value, unit = getattr(self, methodname)(jsondata, key, type_, precision)
  File "/config/custom_components/solis/ginlong_api.py", line 353, in _get_value_from_record
    result = type_(data_raw)
ValueError: invalid literal for int() with base 10: '0.00'

The configuration file is the same as before.

hultenvp, 2022-03-28:
Hey @foncubierta 

I assume the integration does not start up, or does it recover after this failure?

Try to replace the _get_value_from_record method with the code below and see what error then is produced.

```
    def _get_value_from_record(self,
        data: list[dict[str, str]], key: str, type_: type, precision: int = 2
    ) -> tuple[str | int | float | None, str | None]:
        result: str | int | float = None
        unit: str | None = None
        for record in data:
            key_value = record.get('key')
            if key_value == key:
                data_raw = record.get('value')
                if data_raw is not None:
                    try:
                        result = type_(data_raw)
                        # Round to specified precision
                        if type_ is float:
                            result = round(result, precision)
                    except ValueError:
                        _LOGGER.error("Failed to convert %s to type %s, raw value = %s", key, type_, data_raw)
                        value = None
                unit = record.get('unit')
        return result, unit
```

foncubierta, 2022-04-03:
Logger: custom_components.solis.ginlong_api
Source: custom_components/solis/ginlong_api.py:359
Integration: solis ([documentation](https://github.com/hultenvp/solis-sensor/), [issues](https://github.com/hultenvp/solis-sensor/issues))
First occurred: 19:10:20 (1 occurrences)
Last logged: 19:10:20

Failed to convert 3je to type <class 'int'>, raw value = 0.00

hultenvp, 2022-04-04:
Ah, thanks!

May I ask if you have a Sofar inverter with battery? I added this code for these types of devices and it should return the remaining battery capacity in percentage (integer numbers). I'm confused why you see a floating point number.

foncubierta, 2022-04-05:
Indeed, my equipment still does not have batteries, by the way it is a turbo energy inverter and it is compatible with your script, in case you want to add it to the description.

hultenvp, 2022-04-08:
I'm working on a very big change now and will take this along. I'll at least make sure it does no longer hang up discovery.

DdenBraver, 2022-08-07:
The plugin stopped reporting for me on the 5th of august at 13:00 CET.
I tried to reinstall the plugin, and re-enter my configuration but nothing seems to help, it remains "Unavailable"
Does anybody else have this issue?
[image]

commit 322e30b9 (2022-04-08): Add conversion robustness

Fixes #105
closed 2022-08-23 by hultenvp