hultenvp/solis-sensor #106: Yesterday's total energy reported full value next morning, 2 days in row

kazikskrz, 2022-03-29:
I'm having the same here. Every few days I get production value from day before.
version 2.1.0
HA 2022.3.5
[image]
[image]
[image]

_Originally posted by @kazikskrz in https://github.com/hultenvp/solis-sensor/issues/76#issuecomment-1080266134_

kazikskrz, 2022-03-31:
Hello. Enclosing screenshots
[image]
[image]
[image]
[image]
[image]
[image]

hultenvp, 2022-03-31:
Thanks @kazikskrz ,
Inverster status is a deprecated sensor (no longer used). Can you add "solis state" screenshot from same period? (see example)
[image]

kazikskrz, 2022-04-01:
Looks as I did not have this sensor till I have reinstalled integration yesterday. Will paste this State once I get new spike.

hultenvp, 2022-04-01:
It might be a sign your integration was not entirely up to date, causing the spike. 
Chance is high it's gone now.

kazikskrz, 2022-04-04:
Seems like reinstalling integration solved this issue. Thank You.

closed 2022-04-04 by kazikskrz