hultenvp/solis-sensor #108: 2022.3.8 HASS complete failure of this integration

Jonnehs, 2022-04-01:
**Describe the bug**
Invalid config for [sensor.solis]: [inverter_serial] is an invalid option for [sensor.solis]. Check: sensor.solis->inverter_serial. (See ?, line ?).

**Versions**
HA Version 2022.3.8

After updating to 2022.3.8 and v2.2.0 of this integration is seems to have broken the config for this. If you remove the inverter serial line from config.yaml it says the sensors are invalid.

hultenvp, 2022-04-01:
Hi @Jonnehs: Did you read the release notes for this release?
```sensors``` and ```inverter_serial``` config keys are deprecated for a few releases already and have now been removed. Please check that your config is as specified in the README.
If you still experience issues then please add your anonymized config file and debug logs.

hultenvp, 2022-05-07:
Any news, otherwise I close this issue

Jonnehs, 2022-05-07:
Yes sorry, updated. Powerwall installed on Monday so will need to reconfigure it all again...

closed 2022-05-07 by Jonnehs