hultenvp/solis-sensor #110: Local only support

Shadoweee77, 2022-04-11:
Hey,
first of all thank You so much for this!
Would it be possible to make it work local only? Without ever sending data to ginlong. I've stumbled upon this project https://github.com/planetmarshall/solis-service and it looks to me like it's local. Thanks!

hultenvp, 2022-04-14:
Hi @Shadoweee77, yes that would be possible, as a matter of fact the other project I took over maintenance for ([Omnik sensor](https://github.com/hultenvp/home_assistant_omnik_solar)) uses that same principle.
But this integration is focusing on fetching data from the cloud. A local only solution would be a different project outside the scope of this integration.
If you're familiar with python, free to fork the omnik project and replace the omnik-specific decoding with solis coding.

Shadoweee77, 2022-04-14:
Thanks for the reply!
Sadly I have no idea how to use python as of now - learning the general basics atm. Thanks tho!

LucidityCrash, 2022-06-10:
Something that might be worth looking at for people whom it works :
https://github.com/StephanJoubert/home_assistant_solarman/blob/main/README.md

It appers to talk directy to the datalogger and seems to get quite a log of info out.
My Inverter is RHI-(3-6)K-48ES-5G and my data logger has a serial number starting 408 and a firmware version of MW3_15_0501_1.18

hultenvp, 2022-08-23:
Closing as won't fix

closed 2022-08-23 by hultenvp