hultenvp/solis-sensor #112: Can not find the Solis source for inclusion the dashboard

pejannl, 2022-04-26:
Hi. I installed the Wifi plug on the inverter today. I configured it and added my installation to ginlong.com. Data is arriving and I see a correct inverter serial number appearing. So that seems ok.

I installed HACS in Home Assistant (running on a Raspberry) and through HACS I added the solis integration. After a restart I added the parameters in the configuration.yaml and restarted again. That was more than an hour ago.

Note that the configuration is confirmed as valid by Home Assistant.

Then I go to the energy dashboard settings and try to add my solar panels. Unfortunately enough, no Solis source is popping up. I checked the entities list in HAS and there are no solis related entities visible.

What am I doing wrong?

hultenvp, 2022-04-26:
Hey, 

Well that's hard to say. My best guess is that your config contains a small mistake.
Let's start with the config and log. Can you share both here? Please anonymize your config.

If there is nothing in the log then enable debug for the solis inverter. Add the following to the configuration.yaml and restart.

```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```

pejannl, 2022-04-26:
I removed the integration and completely installed it again from scratch. And.... It's working!! Yeah!

closed 2022-04-26 by pejannl