hultenvp/solis-sensor #114: Cant get the plugin to work.

Dduck112, 2022-05-02:
Hello,

First thank you to everybody working on this!! And second Sorry if this is not the place to ask questions. (I'm new to Github and HA)
I have a Solis S6 3600 inverter and want to add it to my home Assistant.
I installed this sensor trough HACS as mentioned in the readme and added the lines to the configuration.yaml but no sensor shows. In the log i get this:
`2022-05-02 07:27:56 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solis/service.py", line 98, in async_discover
    capabilities = await self._do_discover()
  File "/config/custom_components/solis/service.py", line 109, in _do_discover
    if await self._login():
  File "/config/custom_components/solis/service.py", line 87, in _login
    if await self._api.login(async_get_clientsession(self._hass)):
  File "/config/custom_components/solis/ginlong_api.py", line 207, in login
    self._inverter_list = await self.fetch_inverter_list(
  File "/config/custom_components/solis/ginlong_api.py", line 246, in fetch_inverter_list
    for record in result_json['result']['paginationAjax']['data']:
TypeError: string indices must be integers`

I tried to make the portal_plant_id integer and the error is gone, but still no sensor showing. Am I missing something?
from: `portal_plant_id: "1234567"` TO `portal_plant_id: 1234567`
I can login to https://m.ginlong.com/main.html with my email and password and I use them in the configuration.yaml to.
 Full configuration.yaml:  (the / is not in the script but to exclude the form making it a dot)
`
sensor:
  /- platform: solis
    name: "Solis Inverter"
    portal_domain: "m.ginlong.com"
    portal_username: "*********@gmail.com"
    portal_password: "********"
    portal_plant_id: 1234567
`

Any help is very appreciated!! thank you

Dduck112, 2022-05-02:
After some time and uncommenting a debug line 268 in ginlong_api.py   I got this in the log:
`2022-05-02 09:12:48 INFO (MainThread) [custom_components.solis.sensor] Scheduling discovery
2022-05-02 09:12:48 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 1 seconds.
2022-05-02 09:12:57 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!`
but still no sensor to select in the energy dashboard config.

hultenvp, 2022-05-02:
Hey  @Dduck112 

Thanks for the elaborate description, logs and config.

It appears the login sequence fails because the integration receives an unexpected response from the server. It looks like you get a string instead of a JSON payload.
Normally after login the integration requests a list of devices registered with the plant id. This is not working for some reason.

Some things you can try:
- double check your plant id, you already did, but never hurts. In my case it's a 6-digit number. Yours is 7 long?
- try to fully remove and reinstall the integration

If that doesn't work then we'll need to dig further.

Cheers

hultenvp, 2022-05-05:
Hi @Dduck112 ,

Any news?

Dduck112, 2022-05-05:
Hi,

I reinstalled the latest version but no change.
I reinstalled to the previous version but same result. But I do got a other plugin to work whit the local webpage. But I'm still interested in getting this to work to. If I can do any debugging please let me know.
I am going to help the teachers out on a school kamp on Monday till Wednesday so I may not be able to respond quickly.

papacrown, 2022-05-06:
I'm in the same boat as @Dduck112 . I'm getting the following error
```
Logger: homeassistant
Source: custom_components/solis/ginlong_api.py:246 
Integration: solis ([documentation](https://github.com/hultenvp/solis-sensor/), [issues](https://github.com/hultenvp/solis-sensor/issues)) 
First occurred: 4:03:33 PM (1 occurrences) 
Last logged: 4:03:33 PM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solis/service.py", line 98, in async_discover
    capabilities = await self._do_discover()
  File "/config/custom_components/solis/service.py", line 109, in _do_discover
    if await self._login():
  File "/config/custom_components/solis/service.py", line 87, in _login
    if await self._api.login(async_get_clientsession(self._hass)):
  File "/config/custom_components/solis/ginlong_api.py", line 207, in login
    self._inverter_list = await self.fetch_inverter_list(
  File "/config/custom_components/solis/ginlong_api.py", line 246, in fetch_inverter_list
    for record in result_json['result']['paginationAjax']['data']:
TypeError: string indices must be integers
```

And here's my config

```
sensor:
  - platform: solis
    name: My Solis Inverter
    portal_domain: "m.ginlong.com"
    portal_username: "rxxxx@xxxx.com"
    portal_password: "lxxxx"
    portal_plant_id: "198359"
```

I've tried to set all the values with and without the quotation marks, and tried just setting portal plant id without quotations and everything else with, ande vice-versa. Any thoughts? My plant is also available on SolisCloud now, so I also contacted Solis to see if I could move over to that. I used to have this integration working with the inverter serial, but know that has been deprecated for a while.

Thanks!

hultenvp, 2022-05-07:
Thanks both for your input. I'll try to have a look this weekend.

Dduck112, 2022-05-07:
Hi Papacrown,

I use this plugin on hacs now and it works nice.
It can take a while for the first reading to show.

hultenvp, 2022-05-07:
Read [this discussion](https://github.com/hultenvp/solis-sensor/discussions/71#discussioncomment-2538591) if you want to use a local solution.

papacrown, 2022-05-07:
> Read [this discussion](https://github.com/hultenvp/solis-sensor/discussions/71#discussioncomment-2538591) if you want to use a local solution.

I setup the omnik integration for now and it works, but it only creates 2 sensors (energy today and total energy), and I would like a bit more so I'll keep trying with yours.

hultenvp, 2022-05-07:
@Dduck112 , @papacrown,

I just created release 2.2.2. It will dump the server response in debug output so that I can see what the actual response is.
Please post the logging output here or attach it. Not that you need to enable debug for the solis integration:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```

papacrown, 2022-05-07:
got these two log messages, I got the first one if I didn't use quotes around plant id (kept them for username and password), and got both of them if I did use the quotes:

```
This error originated from a custom integration.

Logger: custom_components.solis.ginlong_api
Source: custom_components/solis/ginlong_api.py:252 
Integration: solis (documentation, issues) 
First occurred: 11:03:34 AM (1 occurrences) 
Last logged: 11:03:34 AM

Unknown payload received
```

```
This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/solis/ginlong_api.py:254 
Integration: solis (documentation, issues) 
First occurred: 11:04:01 AM (1 occurrences) 
Last logged: 11:04:01 AM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solis/ginlong_api.py", line 247, in fetch_inverter_list
    for record in result_json['result']['paginationAjax']['data']:
TypeError: string indices must be integers

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/solis/service.py", line 98, in async_discover
    capabilities = await self._do_discover()
  File "/config/custom_components/solis/service.py", line 109, in _do_discover
    if await self._login():
  File "/config/custom_components/solis/service.py", line 87, in _login
    if await self._api.login(async_get_clientsession(self._hass)):
  File "/config/custom_components/solis/ginlong_api.py", line 207, in login
    self._inverter_list = await self.fetch_inverter_list(
  File "/config/custom_components/solis/ginlong_api.py", line 254, in fetch_inverter_list
    slef._online = False
NameError: name 'slef' is not defined

```

hultenvp, 2022-05-07:
Yup, that's me. 'slef' should be 'self'. Has been fixed in v2.2.3
Still I'd have expected debug output, are you certain you've enabled debug in the configuration.yaml?

```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```

papacrown, 2022-05-07:
yeah, this is how I have my logger config

```
# Logging level
logger:
  default: warning
  logs:
    custom_components.solis: debug
```

papacrown, 2022-05-07:
Directly from the full log, sorry I forgot I needed to expand that:

```
2022-05-07 13:51:21 INFO (MainThread) [custom_components.solis.sensor] Scheduling discovery
2022-05-07 13:51:21 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 1 seconds.
2022-05-07 13:51:25 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-05-07 13:51:27 WARNING (MainThread) [custom_components.solis.ginlong_api] Unknown payload received
2022-05-07 13:51:27 DEBUG (MainThread) [custom_components.solis.ginlong_api] {'result': 'OperationPermissionError', 'state': '4'}
2022-05-07 13:51:27 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry.
2022-05-07 13:51:27 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 60 seconds.
```

hultenvp, 2022-05-08:
This is strange.
So you can log in, but the server does not allow you to fetch the details.
Either there's something in your specific server-side settings or there's still an uncovered corner-case in my code and you are actually not logged in.
I'm going to make a version with a lot more debug code, so that we can track the call sequence in more detail and get to the bottom of this.

Can you confirm that the permissionError also occurs in the retries?

papacrown, 2022-05-08:
> This is strange.
>
> So you can log in, but the server does not allow you to fetch the details.
>
> Either there's something in your specific server-side settings or there's still an uncovered corner-case [...]

Yeah that's strange, happy to keep on debugging.... Do you need any info I can get from the ginlong UI?

The permissionError is happening about every minute.

papacrown, 2022-05-08:
And I'm an idiot. My plant I'd was wrong. I had takin it from Solis cloud, not from m.ginlong.com. I'm so sorry for the hassle, this just proves the rule that everything is almost always user error 😂.

hultenvp, 2022-05-09:
🤣. No worries. These kind of issues happen and only make in the end for a more robust integration. I'll add some more improvements to provide better hints in the logging.

@Dduck112 maybe good too check your plant id once more using the info above. If you don't plan on using my integration anymore then I'll close the issue.

Dduck112, 2022-05-09:
I am on s school kamp now. Will check it on Friday. Thanks

hultenvp, 2022-05-17:
Hey @Dduck112 ,

Any news on the issue m

Dduck112, 2022-05-17:
here is my log:

`2022-05-17 11:06:49 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration solis which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-05-17 11:06:49 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-05-17 11:06:49 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration omnik_inverter which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-05-17 11:06:53 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547603657824] Received invalid command: energy/info
2022-05-17 11:06:53 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547603657824] Received invalid command: energy/info
2022-05-17 11:06:53 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547603657824] Received invalid command: energy/info
2022-05-17 11:06:53 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547603657824] Received invalid command: energy/info
2022-05-17 11:06:53 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547603657824] Received invalid command: energy/info
2022-05-17 11:06:57 INFO (MainThread) [custom_components.solis.sensor] Scheduling discovery
2022-05-17 11:06:57 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 1 seconds.
2022-05-17 11:07:11 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error: 
2022-05-17 11:07:11 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry.
2022-05-17 11:07:11 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 60 seconds.
2022-05-17 11:08:21 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error: 
2022-05-17 11:08:21 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry.
2022-05-17 11:08:21 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 60 seconds.
2022-05-17 11:09:31 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error: 
2022-05-17 11:09:31 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry.
2022-05-17 11:09:31 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 60 seconds.
2022-05-17 11:10:35 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-05-17 11:10:37 WARNING (MainThread) [custom_components.solis.ginlong_api] Unknown payload received
2022-05-17 11:10:37 DEBUG (MainThread) [custom_components.solis.ginlong_api] {'result': 'OperationPermissionError', 'state': '4'}
2022-05-17 11:10:37 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry.
2022-05-17 11:10:37 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 60 seconds.`

hultenvp, 2022-05-17:
Hi 

That's you user id :-)
For your plant id please go to the plant tab and there it is. It is explicitly called plant id. 

Hope that helps!

Peter

Dduck112, 2022-05-17:
Hi Peter,

I am a idiot!!
The sensor is nouw showing up in the list at the energie dashboard. thanks.
I do get a view error messages:
[image]

Dduck112, 2022-05-17:
Here is my log now i have changed the plant_id
`2022-05-17 12:05:18 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547898825696] Received invalid command: energy/info
2022-05-17 12:05:18 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547898825696] Received invalid command: energy/info
2022-05-17 12:05:18 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547898825696] Received invalid command: energy/info
2022-05-17 12:05:18 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547898825696] Received invalid command: energy/info
2022-05-17 12:05:18 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547898825696] Received invalid command: energy/info
2022-05-17 12:05:18 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547898825696] Received invalid command: energy/info
2022-05-17 12:05:18 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547898825696] Received invalid command: energy/info
2022-05-17 12:05:18 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547898825696] Received invalid command: energy/info
2022-05-17 12:05:18 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547898825696] Received invalid command: energy/info
2022-05-17 12:05:18 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547898825696] Received invalid command: energy/info
2022-05-17 12:05:18 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547898825696] Received invalid command: energy/info
2022-05-17 12:05:20 INFO (MainThread) [custom_components.solis.sensor] Scheduling discovery
2022-05-17 12:05:20 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 1 seconds.
2022-05-17 12:05:31 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-05-17 12:05:37 DEBUG (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 1001060218120036
2022-05-17 12:05:47 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error: Exception: <class 'asyncio.exceptions.TimeoutError'>
2022-05-17 12:05:47 INFO (MainThread) [custom_components.solis.ginlong_api] Unable to fetch details for device with ID: 104747424
2022-05-17 12:05:47 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry.
2022-05-17 12:05:47 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 60 seconds.`

hultenvp, 2022-05-17:
I see two things in the log:
1) The errors about energy info are typically caused by the system receiving 'none' values, normally they'll go away if the integration settles in.
2) The request for data to ginlong fails with a timeout. That can happen so now and then, especially during startup as CPU load is then big, but if it keeps retrying then something else is wrong. Does it recover after a while and report a successful discovery?

So first discovery must be successful, then the errors of 1) should go away.

hultenvp, 2022-05-21:
Hi @Dduck112 ,

Does it work as expected now?

Dduck112, 2022-05-21:
Hi,
It's not very accurate yet if I compare it to the webpage, but it is working.
Thanks for all the help 😊.

hultenvp, 2022-05-22:
Not sure what energy sensor you use for the energy dash, but energy total has a poor accuracy (1 kWh resolution).  Energy today uses 0.1 kWh resolution. The integration polls every 2 mins ( more is useless as the logger does not send data more frequent to ginlong), so worst case you're 2 mins behind the webpage.

Thanks for the feedback, closing this issue.

closed 2022-05-22 by hultenvp