hultenvp/solis-sensor #117: Ha can't restart after updates.

lex53, 2022-05-15:
Hi,

I'm getting this error and cannot restart.
Any ideas?

Logger: homeassistant.config
Source: config.py:454
First occurred: 10:09:31 PM (1 occurrences)
Last logged: 10:09:31 PM

Invalid config for [sensor.solis]: [inverter_serial] is an invalid option for [sensor.solis]. Check: sensor.solis->inverter_serial. (See /config/sensors.yaml, line 1). Please check the docs at https://github.com/hultenvp/solis-sensor/

hultenvp, 2022-05-16:
Hi @lex53,

You have a [very old config](https://github.com/hultenvp/solis-sensor#ginlong-platform-v2)
inverter_serial is no longer a valid config entry. The integration now autodetects all inverters present in a plant and reports their data fields as sensors.
It has been deprecated for a while and the log produced warnings that the inverter serial was deprecated, so I guess you either skipped a few updates or missed the warnings.
Just remove inverter_serial and your integration will work again.

Cheers.

lex53, 2022-05-16:
Hi thanks for the quick response.

I removed inverter_serial from sensors. Yaml file but it's still not
rebooting. Is that the correct file or should I be looking somewhere else?

Cheers
Alex

hultenvp, 2022-05-16:
Hey Alex

You need to remove it from the configuration.Yaml file.
I assume your sensors.yaml  is included from your configuration.yaml. if so, then that should be fine otherwise edit your configuration.yaml.

As per documentation the config should look as follows:
```
sensor:
  - platform: solis
    name: "My Solis Inverter"
    portal_domain: "m.ginlong.com" (replace for Solarman, Sofar)
    portal_username: "portal_username"
    portal_password: "portal_password"
    portal_plant_id: "plantId"
```

Home assistant can sometimes be hard to convince that the config has changed, but after a reboot it must for sure accept the config.

If still not working then please post the log.

hultenvp, 2022-05-21:
Hi @lex53,

Does it work now?

Cheers

snigehere, 2022-05-21:
Hi.. not so far .. the energy page 
 shows I have generated 34kw today but solis says 17.. still a spike at start of day.. but it did reset just after the spike. 
[image]
[image]

snigehere, 2022-05-21:
Sorry.. looked again and it reset midday still... I need to spend a bit of time to check everything and maybe reset
[image]

hultenvp, 2022-05-26:
Closing due to inactivity from submitter. Feel free to reopen.

closed 2022-05-26 by hultenvp