hultenvp/solis-sensor #118: The value show as "My Solis Inverter Energy Today" is not correctly calculate

snigehere, 2022-05-18:
Thanks for providing this integration and the work you do on refining it.
**Describe the bug**
The value show as "My Solis Inverter Energy Today" is not correctly calculated compared to the data in the solis cloud.

At start of day today I see these numbers in the energy dashboard 

Source	Energy	Cost
My Solis Inverter Energy Today	63.6 kWh	
Solar total	63.6 kWh	
glow_441793525F6C_electric_import	3.3 kWh	£0.14
glow_441793525F6C_electric_export	-2.92 kWh	-£0.22
Grid total	0.38 kWh	-£0.08
glow_441793525F6C_gas_meter	0 m³	
Gas total	0 m³	£0.00

The solar total is wrong - as it was not in the import the energy dashboard assumes I consumed the energy locally and so that means the self generation usage is also wrong. 

Config
sensor:

platform: solis
name: "My Solis Inverter"
portal_domain: "m.ginlong.com"
portal_username: "nxxxx@xxxxxxxxx"
portal_password: "*******"
portal_plant_id: "7********4"

**Versions**
Version	core-2022.5.4
Installation Type	Home Assistant OS

Home Assistant Supervisor
Host Operating System	Home Assistant OS 8.0
Update Channel	stable
Supervisor Version	supervisor-2022.05.2
Docker Version	20.10.14

**Debug logs**
Add to configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**
This picture is a graph of the sensor since I got the integration working a few days ago
 
[image]

On the 16th it looks like the count reset at mid day as the generation went back to zero but on the 17th it did not reset and so the start of the 18th I have the cumulative total, including the prior day, plus part of the day before that, in the daily usage

I end up with over stating the generation for the daily graphs

[image]

This is the solis cloud view of the generation

[image]

This from the energy page in Home Assistant

[image]

hultenvp, 2022-05-19:
Thanks for the elaborate bug report.

'Nice' to see yet another instance of the ginlong-resets-daily-energy-at-the-wrong-moment.

The ginlong portal has the habit of resetting the "energy today" when it comes back online in the morning instead of at midnight. Until that time energy today contains the data of the previous day.
That means I have to ignore any data I get from the portal till I see solis state changing from offline (2) to online (1) again.
To avoid missing a state change back to offline again in the afternoon I limit the filter to AM only.

What I see in your report is that as designed the integration starts reporting energy today again in the morning when (to be confirmed) the inverter comes back online. However the portal is still reporting the energy today from yesterday and continues to do so till noon. Only then it resets.

This is new, until now I've always seen it reset energy today at the same time the inverter comes back online and starts sending data to ginlong again.
It is my gut feeling that the time on the inverter is incorrect and mixes up AM and PM.

Can you check time settings on the inverter and the portal to see if there's something wrong there?
If not, can you give me a screenshot of the "solis state", "solis timestamp inverter online" and "solis timestamp measurements received" sensors on exactly the same timeframe as the "energy today" screenshot in your last posting? 
Also please let me know if you have a normal inverter or a hybrid inverter with battery.

Thanks

snigehere, 2022-05-20:
Thanks for the detailed reply
I will check the inverter timestamp later today.. in the mean time here is the additional info from the app
[image]
[image]
[image]

hultenvp, 2022-05-20:
Thanks for the screenshots.
Those sensors are behaving as expected. So it is really the ginlong server reporting the wrong data because it resets at the wrong moment. 
It always resets at exactly the same moment the inverter comes online again.
Does the data on https://m.ginlong.com/main.html?class=device&subClass=inverter&page=view give the correct Daily Generation (Active)? You can select the parameter in the graph section and then compare to what you see in H.A.

snigehere, 2022-05-20:
Ok.. I have checked the inverter time and you were right .. it was set 12 hours wrong by my installer.  I have corrected that and will wait and see what happens tomorrow

michaelsargent, 2022-05-21:
Just a heads-up.  The servers have been undergoing maintenance the last couple of days.  There was a notice on the ios app.
This happens at China night time.

Regards

Michael

hultenvp, 2022-05-22:
@snigehere, you posted in the wrong issue ;-)
https://github.com/hultenvp/solis-sensor/issues/117#issuecomment-1133604916

It could be that maintenance @michaelsargent is mentioning, but since your the only one seeing this issue I'm still inclining towards a configuration issue. However it resets not exactly at noon, so that might indicate maintenance....
Also check locale settings in ginlong.com (timezone is in plant details) and of your home assistant instance (although I can't see how that one could be wrong)

snigehere, 2022-05-22:
Ok - Thanks the info was still showing up with the reset still happening at mid-day

After digging into the solis web site, as you suspected, I found that the time zone in the "plant info" page on the solis web site (https://m.ginlong.com/main.html) was set to gmt+11 ... I didn't set that so I am guessing that when I created the plant record it picked that up from the time set on the inverter in my home which was set wrong by the installer.  Correcting the time on the physical inverter did not flow through to the solis plant settings.  Seems odd that solid don't take the time from the geo location you set as to where the plant is located on the map.

I used have used SQLite to clear the energy dashboard database and will see what happens tomorrow.  I am fairly confident that this will now be solved.

Part of the issue in diagnosing the problem is that I have two accounts and two plant records with solis - one on https://m.ginlong.com/ and one on https://www.soliscloud.com/ ... the solis cloud is the site that feeds my app on my phone and the graphs on the app looked to be correct.  I set up the plant on https://m.ginlong.com to be able to do the integration in HA and that seems to where the wrong timezone came into play.

I have cleared the energy database out so hopefully I should now get an accurate picture on use vs generation.   If it all works well I will close the issue tomorrow.

closed 2022-05-23 by snigehere

hultenvp, 2022-05-23:
Great to hear and thanks for the update. I'll definitely add it to the readme.
Good luck with the integration.

Cheers