hultenvp/solis-sensor #120: ginlong support denied SolisCloud api key

simbesh, 2022-05-26:
Hello, how have others been able to request an API key from ginlong?

Here is their response to my request:
[image]

any help or tips appreciated, thanks.

hultenvp, 2022-05-26:
Hi, 

Please do not open issues for this, use the [discussion thread](https://github.com/hultenvp/solis-sensor/discussions/71)
The problem you are experiencing is discussed there as well.

Cheers,

closed 2022-05-26 by hultenvp

LucidityCrash, 2022-06-01:
Really have to laugh at this ... put "solis api" into Google the first hit is a link to their own support pages ... you'd think they would know this 😁