hultenvp/solis-sensor #121: Wrong Data in the morning

jahpster, 2022-06-03:
**Describe the bug**
since last week i get wrong data every morning 
I think its the same issue as the other closed bug reports

**Config**

 - platform: solis
    name: "zonnepanelen"
    portal_domain: "m.ginlong.com"
    portal_username: ********
    portal_password: *******
    portal_plant_id: *****

**Versions**
Home Assistant Core 2022.6.1
Home Assistant Supervisor 2022.05.3
HACS version  1.25.5
solis 2.2.3

**Screenshots**
[image]
[image]
[image]

hultenvp, 2022-06-23:
Hi @jahpster,

Apologies, I wrote an answer, but never pressed submit.
Please share energy today history.
It will probably not reset at midnight and show it resets to late. 
The most recent reason why is that the timezone settings in your ginlong.com config are incorrect. So not know why it is an issue now, but it could be one of their recent server updates. Have a look at issue #118 

Cheers

jahpster, 2022-06-27:
Hi,

The timezone is definitely correct in homeassistant and at the plantsettings on ginlong.
The reset happens at midnight from what i can see.

[image]
[image]

hultenvp, 2022-06-27:
I've taken a more close look at your data, I must have rushed through the issue before.
From your post of today I do not see any issues. Everything looks fine, no spikes, nice reset at midnight sharp. Except one thing: your energy today data comes in at a resolution of 1kWh instead of 0.1 kWh.
In your original post it looks exactly the same: too coarse steps. I cannot explain why that is. The integration uses a resolution of 0.1kWh for energy today, so the only thing I can think of is that your logger delivers the data with that resolution or the sensor is actually a different one that got renamed.

If the "yellow spike" in the morning is what's confusing you: that looks like 1kWh production and about the same amount of electricity consumption, so it is registered as settled consumption. This could be coffee machine, toaster, electric warm water heating or combinations. In any case it looks like something in your morning ritual and coincides with first kWh of energy production.

If the 1kWh bugs you, the please check if it is the same on ginlong or not. If ginlong looks fine then maybe a good idea to double check you really are using the energy today sensor, and not another, renamed, sensor.

hultenvp, 2022-10-21:
Closed, no activity

closed 2022-10-21 by hultenvp