hultenvp/solis-sensor #123: No data being collected after a restart

LucidityCrash, 2022-06-08:
A couple of times now after a restart for an upgrade no data has been fetched, this last time I noticed the following in the logs :

 
2022-06-04 10:16:32 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solis/ginlong_base.py", line 70, in __getattr__
    return self._data[name]
KeyError: 'serial'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/solis/service.py", line 175, in async_update
    await self.update_devices(data)
  File "/config/custom_components/solis/service.py", line 131, in update_devices
    serial = getattr(data, INVERTER_SERIAL)
  File "/config/custom_components/solis/ginlong_base.py", line 73, in __getattr__
    raise AttributeError(name) from key_error
AttributeError: serial

after that no data gets fetched you can see this in the history for the timestamp for last data collection : 
[image]

A restart seems to fix things.
And typically I've turned the logging to warning, but I'm thinking this is just a for some reason during startup it is failing to connect to the soliscloud api ?

martin-92, 2022-06-14:
i have the same issue. All the sensors are updating now except the "energy today" sensors.

LucidityCrash, 2022-06-14:
I did an HA upgrade yesterday and got these 
2022-06-13 11:51:51 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry.
2022-06-13 11:53:17 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry.

I didn't pay full attention to when the data started being populated but all looks good just now.
That error implies a timeout when doing the initial connect.

hultenvp, 2022-08-23:
The crash is solved and added robustness + logging to failed discovery.
Please open a new issue if problems persist.
Fixed in https://github.com/hultenvp/solis-sensor/releases/tag/v2.3.1

closed 2022-08-23 by hultenvp