hultenvp/solis-sensor #125: Data stopped being collected after SolisCloud outage

LucidityCrash, 2022-06-18:
Noticed SolisCloud announced an outage on the website for the 17th and it looks like this affected the API (to be expected), got an error in the logs : 
```
2022-06-17 03:26:03 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solis/ginlong_base.py", line 70, in __getattr__
    return self._data[name]
KeyError: 'serial'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/solis/service.py", line 175, in async_update
    await self.update_devices(data)
  File "/config/custom_components/solis/service.py", line 131, in update_devices
    serial = getattr(data, INVERTER_SERIAL)
  File "/config/custom_components/solis/ginlong_base.py", line 73, in __getattr__
    raise AttributeError(name) from key_error
AttributeError: serial
```
and no data has been collected since.  If I run the test script the API is returning data but Home Assistant doesn't seem to be trying to fetch it,

LucidityCrash, 2022-06-19:
As usual a restart of Home Assistant fixed the issue.

hultenvp, 2022-06-21:
I'm going to fix that data objects without serial are allowed, but I'm afraid that only makes the integration more robust, but does not solve the root cause which seems to be that we're receiving incomplete data from the server.

kisjeth, 2022-08-13:
> As usual a restart of Home Assistant fixed the issue.

Not for me. I've try to restart but nothin' change. Same issue, same log.

hultenvp, 2022-08-23:
The crash is solved and added robustness + logging to failed discovery.
Please open a new issue if problems persist.
Fixed in https://github.com/hultenvp/solis-sensor/releases/tag/v2.3.1

closed 2022-08-23 by hultenvp