hultenvp/solis-sensor #126: Solis Energy Today does not reset correctly

bckmrk, 2022-06-21:
**Describe the bug**
Sensor "Energy Today" does not reset correctly and includes yesterdays production in todays total. It has been working fine a couple of months but since 12th of june something has happened. 

**Versions**
Home Assistant Core 2022.6.6
Home Assistant Supervisor 2022.05.3
Home Assistant OS 8.2
HACS Integration version 1.25.5
Frontend version 20220529181005
Integration version v2.2.3

**Debug logs**
Nothing weird in logs? 

2022-06-21 11:34:01 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-06-21 11:34:05 DEBUG (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating inverterstate (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute state for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating timestamponline (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute timestampOnline for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating timestampmeasurement (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute timestampUpdate for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating temperature (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute inverterTemperature for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputvoltagepv1 (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute dcVoltage1 for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputvoltagepv2 (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute dcVoltage2 for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputcurrentpv1 (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute dcCurrent1 for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputcurrentpv2 (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute dcCurrent2 for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputpowerpv1 (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute dcPower1 for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputpowerpv2 (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute dcPower2 for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating acoutputvoltage1 (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute acVoltage1 for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating acoutputvoltage2 (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute acVoltage2 for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating acoutputvoltage3 (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute acVoltage3 for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating acoutputcurrent1 (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute acCurrent1 for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating acoutputcurrent2 (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute acCurrent2 for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating acoutputcurrent3 (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute acCurrent3 for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating actualpower (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute acPower for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating acfrequency (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute acFrequency for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating energylastmonth (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute energyLastMonth for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating energytoday (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute energyToday for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating energythismonth (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute energyThisMonth for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating energythisyear (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute energyThisYear for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.sensor] Creating energytotal (1107A2214150009)
2022-06-21 11:34:07 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute energyTotalLife for inverter 1107A2214150009
2022-06-21 11:34:07 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes.

**Screenshots**
From 1 june the energy today history looks like this: 
[image]

Resent days looks like this: 
[image]

hultenvp, 2022-06-23:
Those people at ginlong are undoubtedly very nice people, but they are not really good at coding. Beats me why you are seeing this and others don't. 

The only thing I could suggest is to double check your timezones. See #118. It's a stretch, because their problem looks different than yours, but I have no other clue what this could be. But as you already figured out yourself this is a server side issue.
Also try power cycling your Inverter.

bckmrk, 2022-06-29:
:-) since the error does not occur every day (as seen in the graphs above, some days are correct without changing anything) I was doubtful to that the time zone change could cure this problem. 

I also looked in my settings to verify that I had set it up correctly and yes, I had the right time zone setting. But I changed from UTC +2 to UTC +1 and now it has worked nice and correct for two days straight. Let’s se what happens next… 🤣

bckmrk, 2022-07-08:
No idea. Nothing happens June 12 so I find it weird that it has something to do with daylight savings… 

still working though. 👍🏼

closed 2022-08-21 by hultenvp