hultenvp/solis-sensor #127: Incorrect system data

ffharaujo, 2022-06-29:
Good afternoon, I am not receiving daily and annual generation data, the inverter temperature value is also not returning the correct information as shown in the image.

If any further information is needed, please let me know.

Thanks for your help

**Versions**
HASS version: Home Assistant Core 2022.6.7
HACS version 1.25.5
Ginlong Solis PV portal integration v2.2.3

**Screenshots**
[image]

ffharaujo, 2022-06-30:
Good afternoon, follow the requested data, please see if it is enough, if you need any more information, please let me know.

Thank you for your help

[Solis.txt](https://github.com/hultenvp/solis-sensor/files/9023185/Solis.txt)

hultenvp, 2022-07-01:
Hey @ffharaujo,

I do not see the server response in the log. Please make sure you uncomment  line 273 in ginlong_api.py?

```
 #_LOGGER.debug("Payload = %s", payload)
```

ffharaujo, 2022-07-03:
Good evening, please check if the information is correct now.

Thank you for your help

[Solis1.txt](https://github.com/hultenvp/solis-sensor/files/9033948/Solis1.txt)

hultenvp, 2022-07-08:
Hi @ffharaujo ,

Finally got around checking the file.
What you seen in the screenshot is what the server returns: It returns 0.0 for energy today data in your case.
From the log, emphasis mine:
{'key': '1bd', 'name': 'Daily Generation (Active)', 'unit': 'kWh', **'value': '0.00'**}

Same holds for inverter temperature: The server sends the info you see in the screenshot:
'key': '1df', 'name': 'Inverter Temperature', 'unit': '℃', '**value': '13.90**'
So the integration is showing the data it gets from the server.

Why this happens is unclear to me.
Some things you could check:
- If you also registered at soliscloud then you might need to re-register your inverter at Ginlong.com to get data again
- Apparently with one of the latest updates of the server they made some changes to localization, meaning wrong data if your timezone is incorrect.

In any case: you'll need to validate the server config

ffharaujo, 2022-07-08:
Perfect, I adjusted the registration on ginlong.com and everything worked perfectly again.

Thanks a lot for the help

hultenvp, 2022-07-08:
Great to hear it works again.

Cheers

closed 2022-07-08 by hultenvp